ros/rosdistro #21312: rosdep update  error

zpzp307, 2019-05-21:
I am a newbie, i uninstalled the ros once , this time the installation is stuct in rosdep.  error:

reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index-v4.yaml
Skip end-of-life distro "ardent"
Add distro "bouncy"
Add distro "crystal"
Add distro "dashing"
Skip end-of-life distro "groovy"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Skip end-of-life distro "jade"
Add distro "kinetic"
ERROR: error loading sources list:
	('The read operation timed out',)

tfoote, 2019-05-22:
This sounds like you have a connectivity issue. 

I you can't resolve it please ask for help on http://answers.ros.org according to our support policy: http://wiki.ros.org/Support We try to keep the issue tracker reserved for bugs and feature developments. There's likely also already some similar questions there that can help you.

closed 2019-05-22 by tfoote

zpzp307, 2019-05-22:
thank you! I try it.

dongpan90, 2021-05-29:
@zpzp307 I met the same problem. How did you solve it?