ros/rosdistro #21513: Noetic Kick-off and Release Checklist

sloretz, 2019-06-11:
This ticket is for things that need to be done for the May 2020 ROS 1 Noetic release. Copied from the [Melodic Release ticket](ros/rosdistro/issues/16139) and [Melodic Kick-Off Checklist](https://github.com/ros/rosdistro/issues/16138).

Prerequisites
---
- [ ] Python 3 support for Melodic checklist (ros/rosdistro#17898)
- [ ] Resolve how to do rosdep keys for python 3

Kickoff part 1
---

- [x] Choose distribution name: **Noetic Ninjemys**
- [ ] Commission a logo
- [ ] Commission a turtlesim icon
- [ ] Add to REP3 with proposed platforms and minimum requirements, request feedback from community
- [ ] List on ROS wiki with expected release date
  * http://wiki.ros.org/Distributions
  * see [example](http://wiki.ros.org/action/diff/Distributions?action=diff&rev1=80&rev2=81)

Ubuntu buildfarm setup
---
- [ ] Add Ubuntu 20.04 to bootstrap repository on repos.ros.org
  * see [example](https://github.com/ros-infrastructure/reprepro-updater/pull/51)
- [ ] Add Ubuntu 20.04  to buildfarm_deployment_config
  * note this is not the live one, it's for reference only
  * see [example](https://github.com/ros-infrastructure/buildfarm_deployment_config/pull/22)
- [ ] Add Ubuntu 20.04 to `ros_buildfarm/common.py`
  * see [example](https://github.com/ros-infrastructure/ros_buildfarm/pull/493)

Python Dependencies setup
---
- [ ] Check that all python packages and scripts below work with python3
  - [ ] [catkin_pkg](https://github.com/ros-infrastructure/catkin_pkg)
  - [ ] [rospkg](https://github.com/ros-infrastructure/rospkg)
  - [ ] [rosdistro](https://github.com/ros-infrastructure/rosdistro)
  - [ ] [rosdep](https://github.com/ros-infrastructure/rosdep)
  - [ ] [vcstools](https://github.com/vcstools/vcstools)
  - [ ] [wstool](https://github.com/vcstools/wstool)
  - [ ] [rosinstall](https://github.com/vcstools/rosinstall)
  - [ ] [bloom](https://github.com/ros-infrastructure/bloom)
  - [ ] [rosinstall_generator](https://github.com/ros-infrastructure/rosinstall_generator)
  - [ ] [osrf_pycommon](https://github.com/osrf/osrf_pycommon)
  - [ ] [catkin_tools](https://github.com/catkin/catkin_tools)
  - [ ] [vcstool](https://github.com/dirk-thomas/vcstool)
  - [ ] [ros_release_python](https://github.com/ros-infrastructure/ros_release_python)
- [ ] Add newly supported platforms to list of platforms for each Python package (in `stdeb.cfg`), in order of dependency
  * do this as soon as debs for new Ubuntu are available, in pre-alpha!
  - [ ] catkin_pkg
    * Example (Melodic: ros-infrastructure/catkin_pkg#198)
  - [ ] rospkg
    * Example (Melodic: ros-infrastructure/rospkg#123)
  - [ ] rosdistro
    * Example (Melodic: ros-infrastructure/rosdistro#108)
  - [ ] rosdep
    * Example (Melodic: ros-infrastructure/rosdep#567)
  - [ ] vcstools
    * Example (Melodic: vcstools/vcstools#127)
  - [ ] wstool
    * Example (Melodic: vcstools/wstool#115)
  - [ ] rosinstall
    * Example (Melodic: vcstools/rosinstall#113)
  - [ ] bloom
    * Example (Melodic: ros-infrastructure/bloom#452)
  - [ ] rosinstall_generator
    * Example (Melodic: ros-infrastructure/rosinstall_generator#42)
  - [ ] osrf_pycommon
    * Example (Melodic: osrf/osrf_pycommon#45)
  - [ ] catkin_tools
    * Example (Melodic: catkin/catkin_tools#497)
  - [ ] vcstool
    * Example(Melodic: dirk-thomas/vcstool@dcee1a0)
[... 119 lines cut ...]
- [ ] update default distro
  * [Example](https://github.com/ros-infrastructure/rosbrowse/pull/8)

**ROS docker images**
- [ ] add Dockerfiles to create ROS docker images

**Misc Infra**
- [ ] Update www.ros.org bottom row widget to point to new release with logo
- [ ] Update www.ros.org/install to point to new release with logo
- [ ] Create Release category on Discourse
  * [Example](https://discourse.ros.org/c/release/melodic)
- [ ] Update mirror documentation
  * [Example](https://github.com/ros-infrastructure/mirror/commits/master) 
- [ ] update ros mirrors
  * [Example](https://github.com/ros-infrastructure/mirror/pull/14)
- [ ] update osrfoundation mirrors
  * Must wait until there are builds for ignition and gazebo for all supported platforms
- [ ] Add Noetic to high level page http://wiki.ros.org/Installation/Ubuntu

**Communication**
- [ ] Announcement pointing to migration guide 
  - [ ] Post to discourse
  - Post to blog
  - Post to G+
  - Post to Twitter

nuclearsandwich, 2019-08-22:
>     * Goal is to make the currently released versions available in the new distros using the `--include` option

The --include option wasn't updated to support the new aptly-based bootstrap repository. I've opened https://github.com/ros-infrastructure/ros_release_python/issues/31 to track support for it and am hoping to tackle it well before it's needed.

SteveMacenski, 2020-03-16:
It looks like `python-catkin-pkg` in Focal is missing from the python file, which blocks `geographic_info`. I can force release it but I feel like that should probably be fixed beforehand.

Thoughts?

sloretz, 2020-03-16:
> It looks like `python-catkin-pkg` in Focal is missing from the python file, which blocks `geographic_info`. I can force release it but I feel like that should probably be fixed beforehand.

@SteveMacenski In ROS Noetic `geographic_info` should depend on the Python 3 equivalent `python3-catkin-pkg` instead. Further, if doesn't need the CLI tools then it should depend on `python-catkin-pkg-modules` in Melodic and earlier, and `python3-catkin-pkg-modules` in Noetic.

SteveMacenski, 2020-03-17:
Ah ok. I'm not a python user in general so I'll probably need someone to sanity check it for me that everything's been properly updated. I did a glance over it and the obvious things I'm aware of from 2.7-3.4 look to be done already.

If that's the case, I can update that stuff and ship it out

clalancette, 2020-07-10:
@sloretz Should we close this out now that Noetic has been released?

clalancette, 2020-09-23:
I'm going to go ahead and close this now that Noetic is released.  Feel free to reopen if needed, or open up follow-up tasks.

closed 2020-09-23 by clalancette