ros/rosdistro #21660: Adding package meta-data to distribution.yaml?

jwhendy, 2019-07-01:
Greetings,

I'm supporting an effort to update ROS on arch. The [distribution.yaml](https://github.com/ros/rosdistro/blob/master/melodic/distribution.yaml) is quite helpful, but I noticed a few things when trying to parse it for our package builds:

- inconsistent data: some packages only have a release url, some have source, sometimes when both are missing, the doc url still points to git (which implies there is a source url)
- sub-packages don't have xml entries: we link to the ROS wiki as the upstream URL in PKGBUILDs; because sub-packages (like `tf2_eigen`) don't have entries in the `release_package_xmls`, I'm not able to know their wiki directly. Currently I'm using a script which sees if `https://wiki.ros.org/pkg_name` exists
- empty data: I'm using ET to find the wiki if it does exist:

```
wiki = ET.fromstring(xmls[pkg]).findall('url[@type="website"]')
```
Sometimes this returns a hit, but it turns out the tag is empty. The `ff` package is an example, and there is a ros wiki page for it...
```
<?xml version="1.0" ?><package><name>ff</name><version>2.1.12</version><description>ff: pddl planner. see http://www.loria.fr/~hoffmanj/ff.html</description><maintainer email="k-okada@jsk.t.u-tokyo.ac.jp">Kei Okada</maintainer><license>GPL</license><url type="website"/><author>J.Hoffmann</author><buildtool_depend>catkin</buildtool_depend><build_depend>mk</build_depend><build_depend>flex</build_depend><build_depend>bison</build_depend><build_depend>unzip</build_depend><build_depend>mk</build_depend><build_depend>rosbuild</build_depend><build_depend>rosbash</build_depend><build_depend>rospack</build_depend><build_depend>roslib</build_depend><build_depend>ca-certificates</build_depend><build_depend>openssl</build_depend><export/></package>
```

Maybe my use-case is kind of fringe, but in principle it would seem to advantageous to have this complete. Any guidance on filling in blanks, standardizing? Interest in tracking sub-packages (e.g. what the package parent is and the wiki page, if it exists)?

clalancette, 2019-07-01:
>     * inconsistent data: some packages only have a release url, some have source, sometimes when both are missing, the doc url still points to git (which implies there is a source url)

This is a feature; sometimes our contributors only want to do one part of the "release" process.  There are separate uses for the documentation, source, and release (binary), and some packages only want parts of it.  So I don't think that we really want to change the flexibility this gives the contributors.

> * sub-packages don't have xml entries: we link to the ROS wiki as the upstream URL in PKGBUILDs; because sub-packages (like `tf2_eigen`) don't have entries in the `release_package_xmls`, I'm not [...]

Sorry, can you point more directly where you are fetching the `release_package_xmls` from?

> 
>     * empty data: I'm using ET to find the wiki if it does exist:
> Sometimes this returns a hit, but it turns out the tag is empty. The `ff` package is an example, and there is a ros wiki page for it...

Oh, I see.  This is a case of the contributor not adding in a `<url>` tag to their package.xml.  In principle, I agree that all packages should have this, but I also don't think we should really enforce it at the rosdistro level.  My suggestion for these packages is to submit a PR to their upstream repository adding the `<url>` tag.

> Interest in tracking sub-packages (e.g. what the package parent is and the wiki page, if it exists)?

Tracking the sub-packages back to the parent is a nice thing.  I would think it would be possible, since our buildfarm/infrastructure code has to do this.  @nuclearsandwich might have some tips on how to achieve this.

jwhendy, 2019-07-01:
Thanks for the reply.

>This is a feature; sometimes our contributors only want to do one part of the "release" process.

I might now know what all these get used for, but it seems kind of arbitrary? Take [bagger](https://github.com/squarerobot/bagger-release) as an example. What benefit is there in not listing [source](https://github.com/squarerobot/bagger), or more particularly listing that page for at least the doc url? I *think* the wiki might rely on some of this, so you get ghost pages as a result. This [isn't helpful](http://wiki.ros.org/bagger), in my opinion.

> can you point more directly where you are fetching the release_package_xmls from?

Whoops! I actually go that from here: http://repositories.ros.org/rosdistro_cache/ (I pulled melodic), which looks to contain distribution.yaml, but indeed has what looks like extracted package.xml data.

> I also don't think we should really enforce it at the rosdistro level.

Good point... I assumed this was maintained, but stupidly should have read contributing.md... now I see the community generates this. Since my script is trying various ways to get the url (is it in the wiki page in the xml data? is there a doc site? does the wiki exist?), I could compile the list of packages that are missing.

>Tracking the sub-packages back to the parent is a nice thing.

I'd be happy to help with that as well. My solutions is certainly low-tech. Do you mean you already have a master index of wiki pages or similar? That's where I ran into trouble. Even with some `time.sleep()` delays, I ran into surge protection checking for package pages...

Thanks for the dialog!

jwhendy, 2019-07-04:
Followup: I now have the list of wiki pages for sub-packages, assuming the `pkg_name` matched up directly with `wiki.ros.org/pck_name`. This was manually scraped and takes some time to not overload the wiki. If there is a better way to get this data, or you have existing code and would like help implementing a more granular listing, I'm game to help!

In addition, I can create a list of missing data for reference and suggest to upstream folks that they add this info to their listings.

tfoote, 2019-07-09:
In general there should be a wiki page for each package name. The wiki pages are not autogenerated but require the maintainer to create them so we don't end up spamming the wiki. 

You'll find that our new site https://index.ros.org does automatically create content for anything indexed. eg [search for ff](https://index.ros.org/search/?term=ff) 

You're right that it's much more helpful for maintainers to submit doc and source entries. But there are cases when it's not appropriate. Doc is not useful if the repository cannot be crawled by our documentation engine. And sometimes the source is not hosted in a format that we support indexing. So we cannot require them. If there are packages which you'd like to see have more content listed please file a ticket with the upstream package to get the attention of the maintainer. 

>> Tracking the sub-packages back to the parent is a nice thing.

> I'd be happy to help with that as well. My solutions is certainly low-tech. Do you mean you already have a master index of wiki pages or similar? That's where I ran into trouble. Even with some time.sleep() delays, I ran into surge protection checking for package pages...

Assuming you're referrring to packages and repositories, (instead of subpackages and packages) index.ros.org already has the associations between those documented. 

If a website url is not declared the wiki page is what we use as the default. It's ok to create a link to it even if it doesn't exist. If someone creates it and there is a documentation job when it's created the header will be populated.

jwhendy, 2019-07-09:
>You'll find that our new site https://index.ros.org does automatically create content for anything indexed. eg search for ff

I have seen that and assumed it was based on distribution.yaml, [risdistro cache](http://repositories.ros.org/rosdistro_cache/), or similar, or that both distribution.yam, and index.ros.org are based on the same thing.

- wiki pages are still missing for some (like [ff](http://wiki.ros.org/ff) even thought it exists)
- index.ros.org indeed tracks parent/child (or repo/packages), but so does `distribution.yaml`, but it doesn't seem to track child wiki pages, e.g. [tf2_eigen](http://wiki.ros.org/tf2_eigen).

This was initially just a neurotic inquiry as I'm helping update the ROS packages for arch linux. We list a url per package in our `PKGBUILD` ([example](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=ros-melodic-tf2-eigen)), and while we were migrating to ROS sources (vs. ros-gbp), I figured I'd update them. That's when I found there appears to be no master list, and many packages listing links.

I think part of the problem is that from an outsider view, I can't tell the difference between accidentally missing (like `ff` having an empty website tag in the distro_cache) and any of the cases you mention above where some link doesn't make sense.

I don't know what other uses having such an index has, but given that I've now built it, it would seem there is no downside in adding that data where possible? Again, happy to help, and I would think it's in everyone's best interest to have these wiki pages listed in various places if they exist...

What if I use the github api to post an issue suggesting a review of the distribution.yaml entry for any packages who had errors? This wouldn't seem to correct the pacakges issue (e.g. tf2_eigen), would it? That would seem to require a new structure and intentionally adding those wiki pages somewhere to the index.

tfoote, 2019-07-12:
> index.ros.org indeed tracks parent/child (or repo/packages), but so does distribution.yaml, but it doesn't seem to track child wiki pages, e.g. tf2_eigen.

I don't understand this there's a link at the right to the wiki page.

> What if I use the github api to post an issue suggesting a review of the distribution.yaml entry for any packages who had errors? 

Can you explain this a little more? Opening a mass of tickets is not going to be helpful. Opening PRs that fix things could be.

> This wouldn't seem to correct the pacakges issue (e.g. tf2_eigen), would it? That would seem to require a new structure and intentionally adding those wiki pages somewhere to the index.

What do you consider wrong with tf2_eigen that needs fixing and could be ticketed? Actually to that matter, what do you want to fix with ff. I see the wiki page has been there since 2015: http://wiki.ros.org/action/info/ff?action=info but you said it's missing above.

jwhendy, 2019-07-12:
>I don't understand this there's a link at the right to the wiki page.

Hmm. Consider me losing my mind. I swear I looked upon the last reply and saw no website link. Now I do. What originally got me on `ff` specifically was the package xml in the distro cache (posted above):

```
<?xml version="1.0" ?><package><name>ff</name><version>2.1.12</version>
...
<url type="website"/>
```

>Opening a mass of tickets is not going to be helpful. Opening PRs that fix things could be.

Agreed.

>What do you consider wrong with tf2_eigen that needs fixing and could be ticketed? 

I actually did not see the wiki link on any packages until you pointed it out, and I thought links to a website, wiki, sources, etc. would live in the main info section.

---

Taking a step back, let me restate how we got here:
- I'm trying to help maintain ROS packages for arch linux
- to do this, we're updating our build files
- to do *that*, I need to know all the packages, their sources, and their repos
- I found this repo, which I believed would contain that
- in parsing the file and the rosdistro_cache, I found things I believed to be missing
- I'm here asking how I can help

I feel friction in asking to do this, so perhaps we should just keep it simple:
- what data source is ros index based on? (if that's the gold standard, I could use that for my aim)
- same question for distribution.yaml and rosdistro_cache
- are all those data sources in agreement with each other? (I believe at least the cache and index do not, based on the missing `ff` url tag)
- if not, do you care?
- if so, would you like any help aligning them?

tfoote, 2019-07-12:
With some slight generalizations this is how the indexing works.

Everything starts with the rosdistro. The distribution.yaml lists all the indexed repositories. To find all the packages you go to those repositories and browse for package.xml files. All data for any given package is pulled from the package.xml. It's expensive to do this walk, so to save most people the cost there's a caching mechanism the rosdistro cache where we store the results of the walk.

This information feeds the wiki, this information feeds the rosindex etc. 

In general there's no duplication of the data except for several layers of caching which can cause temporary inconsistencies, but that's not what you're seeing. You mention that it doesn't seem that all the "sources" are in agreement. However they all are derived from the same source and appear to be in agreement, accurately reproducing the invalid url. 

I definitely do care about the data being correct, but I'm mostly resisting your suggestions about adding or extending the data in the rosdistro which will make things more complicated and cause data redundancy that then could then become inconsistent. In general we rely on the maintainers to keep their package.xml up to date. Looking closer at ff itself, I now see that it's not a missing tag, but an [empty tag](https://github.com/jsk-ros-pkg/jsk_3rdparty/blob/c952692139fc6690854ab8257f8cf33c5b9fdd0b/3rdparty/ff/package.xml#L10) in ff.  And that's an obvious bug with the package and should be ticketed as such on the upstream repository. Once that's updated there it will propagate to all the different places you're seeing like the rosdistro cache, and then the wiki and index.

I'm generally glossing over a lot of details about the exact mechanisms and different paths for different build types, and different paths for different versions. But the main point is that the primary source for all of this information is from the upstream repositories. If there's data that's missing or inaccurate tickets should be opened on the upstream repositories for the attention of the maintainers. If you'd like to suggest better ways to render or display things that are collected by the rosdistro indexing I can point you to the relevant repositories to make suggestions. (Such as [the wiki](https://github.com/ros-infrastructure/roswiki) or [rosindex](https://github.com/ros-infrastructure/rosindex) )

You mention that you don't know if the omissions are purposeful or intentional, and I don't either for any given package. It's a federated community and we delegate that judgement to the package maintainers.

You are right that the rosdistro is your source as the authority for finding the list of packages. I think you'll find that basically everything you need for generating every ROS package is available from the [rosdistro library](http://docs.ros.org/independent/api/rosdistro/html/index.html) and the rosdep database(also in the rosdistro repo). We generate debian packages, ubuntu packages, fedora packages automatically with [bloom](https://wiki.ros.org/bloom) and have support for Gentoo and OpenEmbedded using [superflore](https://github.com/ros-infrastructure/superflore). With those tools you should be able to extend the functionality to generate Arch packages fully automatically as well.

These are also the tools that use the wiki page by default unless the optional website url is set as the package "Homepage" in the debian parlance. in the future we will likely switch that over to use rosindex as the default. The packages that have explicitly declared the default website url will likely need to all be touched. And if you look around you can find ones that are still using our deprecated wiki urls like [bond core](https://github.com/ros/bond_core/blob/393bb134a670e011da41b899dd9d89b20c2fde49/bondpy/package.xml#L13) and they're already out of date. This is why we like to use default values and not type it out every time in all the packages. But if the maintainer wants they can declare a non-standard website like [frame_editor](https://wiki.ros.org/frame_editor) and we add that in parallel since as this example shows the wiki link is still more useful.

nuclearsandwich, 2019-07-12:
> With those tools you should be able to extend the functionality to generate Arch packages fully automatically as well.

Work has also been done in the past to support adding an Archlinux generator to Bloom and work is ongoing to reduce the friction and maintenance burden of supporting additional generators with the hope that Superflore's separate generators are someday merged into Bloom so that Superflore uses Bloom under the hood.

jwhendy, 2019-09-03:
@tfoote Thanks for the in-depth explanation! Forgive the delay. I had some vacation, house work, and also just ran out of time to work on this for a bit. I appreciate the understanding of package details as well as the mention of `bloom` and `superflore`, which are definitely of interest re. building arch binaries.

I [asked](https://github.com/jsk-ros-pkg/jsk_3rdparty/issues/174) that they add a website to `ff` (which will likely point to the ROS wiki from looking at their other package.xml files).

Re. bloom and superflore, I'll have to look into these more and they seem above my head at the moment. Bloom tells me how to add a package (which presumably builds for supported ROS distros), but I think what we need is that existing packages be build for arch, specifically.

superflore looks like the equivalent of an AUR package manager, namely "grab all the required source files for packages x, y, and z, then build and install them on my system." If so, I think that's basically what arch users already do. I think we're hung up on finding a build farm to do this (and then creating the infrastructure/config required for arch). I'm just helping out the lead of this effort who sounds to have more experience on that front.

Thanks again!

closed 2019-09-03 by jwhendy