ros/rosdistro #21718: RFC: Update Install instructions not to use SKS servers

tfoote, 2019-07-08:
There's some significant issues with them recently.

Re: https://gist.github.com/rjhansen/67ab921ffb4084c865b3618d6955275f

We should also look to see what debian is doing about it. @j-rivero @nuclearsandwich have you seen any traffic on this?

j-rivero, 2019-07-12:
> We should also look to see what debian is doing about it. @j-rivero @nuclearsandwich have you seen any traffic on this?

I did not see anything in the debian mailing lists. The friends of Gentoo [released a note](https://www.gentoo.org/news/2019/07/03/sks-key-poisoning.html) about the issue and how they protected their keyring from this kind of attack.

A part from this, I've only seen mitigation measures at user system level.

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas