ros/rosdistro #22075: NumPy, SciPy, Pandas, etc. require Python >= 3.5

MatthijsBurgh, 2019-08-22:
The newest releases of NumPy, Scipy, Pandas, etc. require Python >= 3.5. When installing in example python-keras-pip with rosdep, the newest version of SciPy is installed. This fails because Python 2.7 is used

This causes our CI to fail, which uses industrial CI. 

I have not been able to come up with a generic solution for this.

moriarty, 2019-08-22:
https://pythonclock.org/ :bomb: :boom:

moriarty, 2019-08-22:
I asked @MatthijsBurgh which repo this was happening for:

https://github.com/tue-robotics/image_recognition/blob/master/image_recognition_keras/package.xml

moriarty, 2019-08-23:
Dockerfile to reproduce:

```Dockerfile
ARG ROS_VERSION=kinetic
FROM ros:$ROS_VERSION

ENV ROS_DISRO=$ROS_VERSION

RUN DEBIAN_FRONTEND=noninteractive apt-get update \
 && apt-get install -y -q --no-install-recommends \
    python-catkin-tools \
    python-pip \
    python-rosinstall-generator \
 && . /opt/ros/$ROS_DISTRO/setup.sh \
 && mkdir -p $HOME/ros_ws/src \
 && cd $HOME/ros_ws \
 && git clone https://github.com/tue-robotics/image_recognition $HOME/ros_ws/src/tue-robotics \
 && catkin config --init \
 && catkin config --install --extend /opt/ros/$ROS_DISTRO \
 && rosdep install --from-paths src --ignore-src -r -y
```

Some error output
```
Collecting keras
  Downloading https://files.pythonhosted.org/packages/f8/ba/2d058dcf1b85b9c212cc58264c98a4a7dd92c989b798823cc5690d062bb2/Keras-2.2.5-py2.py3-none-any.whl (336kB)
Collecting scipy>=0.14 (from keras)
  Downloading https://files.pythonhosted.org/packages/ee/5b/5afcd1c46f97b3c2ac3489dbc95d6ca28eacf8e3634e51f495da68d97f0f/scipy-1.3.1.tar.gz (23.6MB)
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-7eTw4o/scipy/setup.py", line 31, in <module>
        raise RuntimeError("Python version >= 3.5 required.")
    RuntimeError: Python version >= 3.5 required.
    
    ----------------------------------------
```

```
ERROR: the following rosdeps failed to install
  pip: command [pip install -U tensorflow] failed
  pip: Failed to detect successful installation of [tensorflow]
  pip: command [pip install -U keras] failed
  pip: Failed to detect successful installation of [keras]
executing command [apt-get install -y python-matplotlib]
executing command [apt-get install -y python-catkin-lint]
executing command [pip install -U tensorflow]
executing command [apt-get install -y ros-kinetic-rqt-gui]
executing command [apt-get install -y ros-kinetic-image-transport]
executing command [apt-get install -y python-scipy]
executing command [apt-get install -y ros-kinetic-diagnostic-updater]
executing command [pip install -U keras]
executing command [apt-get install -y python-opencv]
executing command [apt-get install -y ros-kinetic-rqt-gui-py]
executing command [apt-get install -y ros-kinetic-cv-bridge]
executing command [apt-get install -y python-sklearn]
The command '/bin/sh -c DEBIAN_FRONTEND=noninteractive apt-get update  && apt-get install -y -q --no-install-recommends     python-catkin-tools     python-pip     python-rosinstall-generator  && . /opt/ros/$ROS_DISTRO/setup.sh  && mkdir -p $HOME/ros_ws/src  && cd $HOME/ros_ws  && git clone https://github.com/tue-robotics/image_recognition $HOME/ros_ws/src/tue-robotics  && catkin config --init  && catkin config --install --extend /opt/ros/$ROS_DISTRO  && rosdep install --from-paths src --ignore-src -r -y' returned a non-zero code: 1
```

## FIX: upgrade pip

```Dockerfile
ARG ROS_VERSION=kinetic
FROM ros:$ROS_VERSION

ENV ROS_DISRO=$ROS_VERSION

RUN DEBIAN_FRONTEND=noninteractive apt-get update \
 && apt-get install -y -q --no-install-recommends \
    python-catkin-tools \
    python-pip \
    python-rosinstall-generator \
 && pip install --upgrade pip \
 && hash -r pip \
 && . /opt/ros/$ROS_DISTRO/setup.sh \
 && mkdir -p $HOME/ros_ws/src \
 && cd $HOME/ros_ws \
 && git clone https://github.com/tue-robotics/image_recognition $HOME/ros_ws/src/tue-robotics \
 && catkin config --init \
 && catkin config --install --extend /opt/ros/$ROS_DISTRO \
 && rosdep install --from-paths src --ignore-src -r -y
```

```
executing command [apt-get install -y python-matplotlib]
executing command [apt-get install -y python-catkin-lint]
executing command [pip install -U tensorflow]
executing command [apt-get install -y ros-kinetic-rqt-gui]
executing command [apt-get install -y ros-kinetic-image-transport]
executing command [apt-get install -y python-scipy]
executing command [apt-get install -y ros-kinetic-diagnostic-updater]
executing command [pip install -U keras]
executing command [apt-get install -y python-opencv]
executing command [apt-get install -y ros-kinetic-rqt-gui-py]
executing command [apt-get install -y ros-kinetic-cv-bridge]
executing command [apt-get install -y python-sklearn]
#All required rosdeps installed successfully
Removing intermediate container b1e407c87c8f
 ---> 2ccbd757a1f1
Successfully built 2ccbd757a1f1
```

```
Installing collected packages: keras
Successfully installed keras-2.2.5
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
```

MatthijsBurgh, 2019-08-23:
Upgrading pip does solve the issue. Thanks for your help @moriarty.

closed 2019-08-23 by MatthijsBurgh