ros/rosdistro #23193: libqglviewer-dev-qt4 on armhf

wxmerkt, 2019-12-02:
octovis on dashing armhf fails as `libqglviewer-dev-qt4` cannot be resolved. This appears to be missing upstream. Can we blacklist the package then for armhf or what is the correct workaround?

tfoote, 2019-12-17:
yeah, if a dependency like that is missing for a specific arch blacklisting it is our usual approach. We generally like to open an upstream ticket for reference and then open a PR blacklisting it so that it's visible why the package has disappeared. As an example: https://github.com/ros-infrastructure/ros_buildfarm_config/pull/126 where they resolve it and we reverted the blacklist.

closed 2019-12-17 by tfoote

wxmerkt, 2019-12-17:
Thank you for the tutorial @tfoote.

I have opened pull requests for dashing (https://github.com/ros2/ros_buildfarm_config/pull/55) and eloquent (https://github.com/ros2/ros_buildfarm_config/pull/55).