ros/rosdistro #23341: Can't find dependency although existing in yaml-file.

tylernewnoise, 2019-12-19:
On Ubuntu 16.04 I tried to install yapf via rosdep but rosdep can't find the package although it exists in [python.yaml](https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml) (line 5754). Although it works if I change the order of the yaml files in `/etc/ros/rosdep/sources.list.d/20-default.list`. 
Please note that yapf also exits in [base.yaml](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml), but not for Ubuntu Xenial. I guess the search-algorithm for the package stops too early.

tfoote, 2019-12-31:
The rules are explicitly evaluated in order and go based on the first match to allow custom overrides of the values on system administrators or users.

It looks like #22073 was added without consolidating with #15567 

These should be consolidated to leverage the pip values for older platforms and binary packages where available. @tylernewnoise Could you open a PR to deduplicate the entries? I'd recommend explicitly listing the older platforms with pip entries, and anywhere they're availabe from binary packages with the wildcard notation to allow the rule to match expected future distributions including the binary packages again.

tylernewnoise, 2020-01-06:
Hi and a happy new year.

Thx for pointing it out - I'll make the PR asap.

PR #23414 (2020-01-08): Deduplicate yapf entry (fix for issue #23341)
This PR is intended to fix #23341.
closed 2020-01-08 by clalancette