ros/rosdistro #24016: Failing of iirob_filters package for debian stretch

destogl, 2020-03-10:
HI all,

my package KITrobotics/iirob_filters is failing for [melodic on debian stretch](http://build.ros.org/job/Mbin_ds_dS64__iirob_filters__debian_stretch_amd64__binary/29/). The error seams to be the following:
```
 undefined reference to `void tf2::doTransform<geometry_msgs::Vector3Stamped_<std::allocator<void> > >(geometry_msgs::Vector3Stamped_<std::allocator<void> > const&, geometry_msgs::Vector3Stamped_<std::allocator<void> >&, geometry_msgs::TransformStamped_<std::allocator<void> > const&)'
collect2: error: ld returned 1 exit status
```

Does anyone have an idea why is this?

clalancette, 2020-09-01:
> Does anyone have an idea why is this?

I'm sorry for the very long delay in answering here.

Looking at the iirob_filters code, it looks like https://github.com/KITrobotics/iirob_filters/blob/12cae183ab30200ff702baf4f0ef57e625e3c532/include/iirob_filters/gravity_compensation.h#L167 is the line that is probably failing.  `tf2::doTransform` is a templated version in https://github.com/ros/geometry2/blob/dae26a225e6c14a5aabd25a8b08c3031e6563535/tf2_geometry_msgs/include/tf2_geometry_msgs/tf2_geometry_msgs.h#L978

So it looks like you need to do a few things:
1.  Make sure the package.xml depends on tf2_geometry_msgs
1.  Make sure to add `tf2_geometry_msgs` to the find_package call in your CMakeLists.txt
1.  Make sure to `#include <tf2_geometry_msgs/tf2_geometry_msgs.h>` in your gravity_compensation.h file.

I *think* if you do all of that, it will compile properly (and that is the more correct thing to do for Ubuntu as well).

Let me know if this helps.

destogl, 2020-09-15:
Should be corrected in #26561 and #26562

clalancette, 2020-09-23:
I'm closing this out, as I believe this is solved.  Feel free to reopen if needed.

closed 2020-09-23 by clalancette