ros/rosdistro #24251: Import catkin_lint 1.6.7

roehling, 2020-03-27:
It has been suggested to me that I would be nice if a more recent version of `catkin_lint` were available in the OSRF ROS repository, since `industrial_ci` prefers to setup its testing environment from there.

AFAICT the `ros-infrastructure/reprepro-updater` still has a valid configuration to fetch the package from my PPA.

gavanderhoorn, 2020-03-27:
> since `industrial_ci` prefers to setup its testing environment from there.

This is not just about `industrial_ci` I would say.

As @roehling writes in https://github.com/UniversalRobots/Universal_Robots_ROS_Driver/pull/132#issuecomment-605055995:

> Given that catkin_lint has become part of Debian upstream and the latest release is always available from PyPI and my PPA, I felt that there's no need to bother OSRF any more.

ROS Answers is full of comments and answers discouraging people from using `pip` to install anything ROS related. It's true there are valid reasons for using `pip`, but the general sentiment is "pip == bad", especially on Debian/Ubuntu.

The PPA is available (:100: @roehling), but unless users look for it, it won't be added to their systems.

Everyone installing ROS will have `packages.ros.org` configured, so whenever they run `apt install python-catkin-lint`, they will get the version from either that, or the one which happens to be available from upstream.

In many cases, this means installing an old version of `catkin_lint`.

It would be great if we could import the newer releases of `catkin_lint` to `packages.ros.org`, as that would avoid this.

tfoote, 2020-03-27:
In general yes, we do not recommend using the pip installation methods in conjunction with any apt methods. It's really easy to accidentally overlay the apt installation and then get incompatible versions that are not tracked. It often will come back to bite you later when apt automatically updates the system but the pip version is still at the overridden state. 

On the flip side in general our policy is that once a package is available upstream we don't import or override it because that will potentially break anybody else who's using or reusing the upstream packages and then also wants to use the ROS sources. With it upstream already we can think of importing newer versions as backports, so we can consider it under that framework. And if it's backwards compatible it's something we can then consider continuing to import. But in general we're working to reduce our number of bootstrap imports.

Note that we're not using the reprepro for the boostrap repository anymore so although that config exists we'll need to reproduce it for aptly.

@nuclearsandwich FYI

gavanderhoorn, 2020-03-28:
> On the flip side in general our policy is that once a package is available upstream we don't import or override it because that will potentially break anybody else who's using or reusing the [...]

just a comment here: `catkin_lint` is an almost stand-alone Q&A tool, and the disadvantage of depending on upstream to provide it is the fact that upstream is almost inherently out-of-date. For a Q&A tool, that's deadly.

I fully appreciate the maintenance overhead and difficulties with libraries like PCL, OpenCV and similar pieces of software though.

gavanderhoorn, 2020-08-31:
@nuclearsandwich: would it be possible to import from the new ROS PPA @roehling mentions?

nuclearsandwich, 2020-10-09:
@roehling does https://github.com/ros-infrastructure/reprepro-updater/pull/81 look correct to you?
I'm trying to get a little bit more standardized on how we both select and advertise the upstreams for the ROS repository. When there's a new version of catkin-lint a PR updating the version in that config will let us know there's a new version to pull.

roehling, 2020-10-21:
I've noticed that the updater ran successfully and updated the [Main repository in repositories.ros.org](http://repositories.ros.org/ubuntu/main/pool/main/c/catkin-lint/), but that update did not propagate to [packages.ros.org](http://packages.ros.org/ros/ubuntu/pool/main/c/catkin-lint/) yet. Is this intentional, or is there a bug in the update script?

dirk-thomas, 2020-10-27:
@nuclearsandwich Friendly ping.

nuclearsandwich, 2020-10-28:
The upload_main to trigger a sync of the primary mirrors with the build farm repository wasn't triggered and we haven't yet had a sync on that farm that would trigger it otherwise. I've manually triggered http://build.ros.org/job/upload_main/108/ and that should rectify the issue.

nuclearsandwich, 2020-10-31:
Since the attempted fix in my last message catkin-lint 1.6.10 is now on packages.ros.org I'm going to close the issue but if there's some problem please feel free to re-open or open a new issue. If the catkin-lint package updates and that update should be imported into the ROS repositories send a PR to reprepro-updater like this one https://github.com/ros-infrastructure/reprepro-updater/pull/81 with the updated version.

closed 2020-10-31 by nuclearsandwich