ros/rosdistro #24377: Github actions for this repo run on forks and fail

sloretz, 2020-04-06:
Reported by @mikeferguson https://github.com/ros/rosdistro/pull/24364#issuecomment-609761301

> any idea why the tests are also running (and failing) on my account?https://github.com/mikeferguson/rosdistro/actions Seems like maybe the config for the new actions-based testing is not right?

It looks the actions for PR jobs are also running on forks, but failing. There seems to be a lot of console log spam of complaining about `ROSGHPRB_TOKEN`, which sounds like it might be an environment variable with an API key. Is there a way to disable actions on forks of rosdistro, or to bail early if the token is not set?

```
No ROSGHPRB_TOKEN set, continuing without checking hooks
No ROSGHPRB_TOKEN set, continuing without checking hooks
No ROSGHPRB_TOKEN set, continuing without checking hooks
No ROSGHPRB_TOKEN set, continuing without checking hooks
No ROSGHPRB_TOKEN set, continuing without checking hooks
No ROSGHPRB_TOKEN set, continuing without checking hooks
```

dirk-thomas, 2020-04-06:
The actual error was a traceback for me:

```
Traceback (most recent call last):
  File "/home/runner/work/rosdistro/rosdistro/test/test_url_validity.py", line 328, in test_function
    detected_errors = main()
  File "/home/runner/work/rosdistro/rosdistro/test/test_url_validity.py", line 311, in main
    errors = check_repo_for_errors(r)
  File "/home/runner/work/rosdistro/rosdistro/test/test_url_validity.py", line 192, in check_repo_for_errors
    source_errors = check_source_repo_entry_for_errors(repo['source'], tags_and_commits_valid, tags_and_commits_valid)
  File "/home/runner/work/rosdistro/rosdistro/test/test_url_validity.py", line 156, in check_source_repo_entry_for_errors
    version = source['version'] if source['version'] else None
KeyError: 'version'
```

tfoote, 2020-04-06:
Interesting that the actions config propogates to the forks unlike travis. It looks like the failure is due to it diffing to origin/master where if your origin is not up to date the diff will be too big. I need to think about how to cleanly force it to check against upstream master.

closed 2020-04-07 by tfoote

dirk-thomas, 2020-05-20:
@tfoote I had another action run and fail on my fork: https://github.com/dirk-thomas/rosdistro/actions/runs/110643817

commit 556bff8d (2020-04-07): Setup CI to explicitly fetch upstream not just use master which may be out of date on forks.
Fixes #24377