ros/rosdistro #24465: Specify a version of a pip library

asmsuechan, 2020-04-14:
Hello, developers.

I have a new ROS package written in Python and am trying to upload it to apt via this repository.

My issue is the version specification of a pip library.

my package does not work well on the latest version of a dependent package. Therefore, I have to specify a version on `pip install`.

I know `rosdep/python.yaml` exists for solving ROS package dependencies. However, I suppose I can not specify the version of a pip library at the file and the file only to specify the latest version of a pip library.

I would happy I can run `pip install -r requirements.txt` on ROS package installation. Is this feasible? Or, are there any solutions?

asmsuechan, 2020-04-14:
I suppose I should ask at https://answers.ros.org/ rather than here.

closed 2020-04-14 by asmsuechan

Timple, 2021-02-16:
Link to question because this is the first google result:
https://answers.ros.org/question/349374/rosdep-how-to-specify-a-version-of-a-pip-library-on-install/