ros/rosdistro #25293: Wrong version of libpcl in Debian Buster

lepalom, 2020-06-03:
base.yaml has the entry:

libpcl-all:
  debian:
    buster: [libpcl-apps1.8, libpcl-common1.8, libpcl-features1.8, libpcl-filters1.8, libpcl-io1.8, libpcl-kdtree1.8, libpcl-keypoints1.8, libpcl-ml1.8, libpcl-octree1.8, libpcl-outofcore1.8, libpcl-people1.8, libpcl-recognition1.8, libpcl-registration1.8, libpcl-sample-consensus1.8, libpcl-search1.8, libpcl-segmentation1.8, libpcl-stereo1.8, libpcl-surface1.8, libpcl-tracking1.8, libpcl-visualization1.8]

The correct value is 1.9 not 1.8 see [pcl in Debian Buster](https://packages.debian.org/source/buster/pcl).

I don't know if this is the cause because some packages that depends on libpcl in Noetic could not be installed.

clalancette, 2020-06-03:
Yep, looks like you are right and this is a bug.  Can you contribute a fix?  Thanks.

lepalom, 2020-06-03:
What do you want exactly? Can I put the correct line here or should I make a PR? It's just a number.

clalancette, 2020-06-03:
Make a PR, please.  Make sure to link to the Debian Buster database as you did above, and say in the PR that it fixes this bug (#25293).  Thanks.

lepalom, 2020-06-03:
Well, let me check how to do it. i have found 61 packages with the entry debian or buster that are not in Buster. I have to check edit manually to make a complete PR.

dirk-thomas, 2020-09-01:
@ lepalom Friendly ping.

lepalom, 2020-09-01:
Yes, I forgot that!!!

I'm sorry. I check my notes to find the packages and try to make the PR.

PR #26487 (2020-09-10): Fix libpcl-all entry for Debian buster. Fix #25293
Fix libpcl-all entry for Debian Buster. Fix #25293
closed 2020-09-10 by ivanpauno