ros/rosdistro #25334: restore folding logic from Travis or update error messages

dirk-thomas, 2020-06-04:
See e.g. https://github.com/ros/rosdistro/blob/a8d19c8bd7db8cf720da53882959ac4a5bb53482/test/rosdistro_check_urls_test.py#L22-L29

Regression from #24333.

dirk-thomas, 2020-06-18:
@tfoote Friendly ping.

tfoote, 2020-06-19:
This was discussed in the review on that as an acceptable regression with the improved performance. I'm not going to have time to look at it right now. 

If someone has time to look at it . There looks like there's a group and ungroup is an alternative to folding:
- https://github.community/t/has-github-action-somthing-like-travis-fold/16841
- https://github.community/t/travis-fold-equivalent/18256
- https://github.com/actions/toolkit/blob/df7e2c13c89ea93069cad62f1adfe78a87111cc0/docs/commands.md#group-and-ungroup-log-lines

closed 2026-10-29 by mjcarroll