ros/rosdistro #25392: Error while releasing ros package: Remote branch not found in upstream

pankhurivanjani, 2020-06-10:
Hello!
I am trying to release a package in ros melodic whose PR has been accepted [jderobot_color_tuner](https://github.com/ros/rosdistro/pull/25367). 

While submitting the PR, I received an email from rosbuildfarm stating 'Jenkins build is back to normal',  here is the report of success of build [jenkins-success-report](http://build.ros.org/job/Mdev__jderobot_color_tuner__ubuntu_bionic_amd64/4/console)

But, after my PR was merged yesterday, I received the email for build failure. The report is attached here [console_report_build_failure](http://build.ros.org/job/Msrc_uB__jderobot_color_tuner__ubuntu_bionic__source/7/console#console-section-10)

Primarily, I am getting error in the part
"10. Run Dockerfile - generate sourcedeb ->get sources", the logs for the same are:

```
08:00:00 # BEGIN SUBSECTION: get sources
08:00:03 Invoking 'git clone --branch debian/ros-melodic-jderobot-color-tuner_0.0.4-2_bionic --depth 1 --no-single-branch https://github.com/JdeRobot/ColorTuner-release.git /tmp/sourcedeb/source'
08:00:03 Cloning into '/tmp/sourcedeb/source'...
08:00:05 fatal: Remote branch debian/ros-melodic-jderobot-color-tuner_0.0.4-2_bionic not found in upstream origin
08:00:05 # END SUBSECTION
08:00:05 Traceback (most recent call last):
08:00:05   File "/tmp/ros_buildfarm/scripts/release/get_sources.py", line 50, in <module>
08:00:05     sys.exit(main())
08:00:05   File "/tmp/ros_buildfarm/scripts/release/get_sources.py", line 46, in main
08:00:05     args.os_name, args.os_code_name, args.source_dir, args.debian_repository_urls)
08:00:05   File "/tmp/ros_buildfarm/ros_buildfarm/sourcedeb_job.py", line 52, in get_sources
08:00:05     subprocess.check_call(cmd)
08:00:05   File "/usr/lib/python3.6/subprocess.py", line 311, in check_call
08:00:05     raise CalledProcessError(retcode, cmd)
08:00:05 subprocess.CalledProcessError: Command '['git', 'clone', '--branch', 'debian/ros-melodic-jderobot-color-tuner_0.0.4-2_bionic', '--depth', '1', '--no-single-branch', 'https://github.com/JdeRobot/ColorTuner-release.git', '/tmp/sourcedeb/source']' returned non-zero exit status 128.
08:00:06 Build step 'Execute shell' marked build as failure
08:00:06 $ ssh-agent -k
08:00:06 unset SSH_AUTH_SOCK;
08:00:06 unset SSH_AGENT_PID;
08:00:06 echo Agent pid 7747 killed;
08:00:06 [ssh-agent] Stopped.
08:00:07 [description-setter] Could not determine description.
08:00:07 Sending e-mails to: clalancette+buildfarm@osrfoundation.org ros-buildfarm-melodic@googlegroups.com pankhurivanjani@gmail.com
08:00:08 'Disable Failed Job Plugin' Disabled Job
08:00:08 Finished: FAILURE
```

I followed the regular procedure of release using bloom and as states in the error I can't see the branch 'debian/ros-melodic-jderobot-color-tuner_0.0.4-2_bionic' in 'https://github.com/JdeRobot/ColorTuner-release.git. 

@clalancette  It would be really helpful if you could suggest something for this. I fixed many errors following your suggestions in the last discussion over email but I am unable to get through this part.

Thanks,

clalancette, 2020-06-12:
So, as you can see looking in the release repository, that particular branch is missing.  The only thing I can think there is that bloom was not successful in resolving the rosdep keys for debian, and so it didn't generate the branches properly.  In that case, it *should* have thrown an error, but I'm not sure.

Can you do another bloom-release on the package and attach all of the output here?  That may give us a clue of what is going on during the release process.

pankhurivanjani, 2020-06-20:
Hello!  Thanks for your response, Sorry I got engaged in some examination stuff and getting back to it now. 

I release again and got no errors, here is the output:

```
pankhuri@pankhuri-LENOVO-IDEAPAD-500-15ISK:~/ColorTuner/jderobot_color_tuner$ bloom-release --rosdistro melodic --track melodic jderobot_color_tuner --edit
ROS Distro index file associate with commit '5a384b4bc7fdb0fa0a5ab211a42bcc63a5964e5d'
New ROS Distro index url: 'https://raw.githubusercontent.com/ros/rosdistro/5a384b4bc7fdb0fa0a5ab211a42bcc63a5964e5d/index-v4.yaml'
==> Fetching 'jderobot_color_tuner' repository from 'https://github.com/JdeRobot/ColorTuner-release.git'
Cloning into '/tmp/tmpR0KCFW'...
remote: Enumerating objects: 1186, done.
remote: Counting objects: 100% (1186/1186), done.
remote: Compressing objects: 100% (851/851), done.
remote: Total 1186 (delta 340), reused 1104 (delta 262), pack-reused 0
Receiving objects: 100% (1186/1186), 11.75 MiB | 3.81 MiB/s, done.
Resolving deltas: 100% (340/340), done.
Track 'melodic' exists, editing...
Repository Name:
  upstream
    Default value, leave this as upstream if you are unsure
  <name>
    Name of the repository (used in the archive name)
  ['jderobot_color_tuner']: 
Upstream Repository URI:
  <uri>
    Any valid URI. This variable can be templated, for example an svn url
    can be templated as such: "https://svn.foo.com/foo/tags/foo-:{version}"
    where the :{version} token will be replaced with the version for this release.
  ['https://github.com/JdeRobot/ColorTuner.git']: 
Upstream VCS Type:
  svn
    Upstream URI is a svn repository
  git
    Upstream URI is a git repository
  hg
    Upstream URI is a hg repository
  tar
    Upstream URI is a tarball
  ['git']: 
Version:
  :{ask}
    This means that the user will be prompted for the version each release.
    This also means that the upstream devel will be ignored.
  :{auto}
    This means the version will be guessed from the devel branch.
    This means that the devel branch must be set, the devel branch must exist,
    and there must be a valid package.xml in the upstream devel branch.
  <version>
    This will be the version used.
    It must be updated for each new upstream version.
  [':{auto}']: 
Release Tag:
  :{none}
    For svn and tar only you can set the release tag to :{none}, so that
    it is ignored.  For svn this means no revision number is used.
  :{ask}
    This means the user will be prompted for the release tag on each release.
  :{version}
    This means that the release tag will match the :{version} tag.
    This can be further templated, for example: "foo-:{version}" or "v:{version}"
    
    This can describe any vcs reference. For git that means {tag, branch, hash},
    for hg that means {tag, branch, hash}, for svn that means a revision number.
    For tar this value doubles as the sub directory (if the repository is
    in foo/ of the tar ball, putting foo here will cause the contents of
    foo/ to be imported to upstream instead of foo itself).
  [':{version}']: 
Upstream Devel Branch:
  <vcs reference>
    Branch in upstream repository on which to search for the version.
[... 286 lines cut ...]
From https://github.com/ros/rosdistro
 * branch                master     -> FETCH_HEAD
==> git reset --hard 5a384b4bc7fdb0fa0a5ab211a42bcc63a5964e5d
HEAD is now at 5a384b4bc soem: 1.4.1001-1 in 'noetic/distribution.yaml' [bloom] (#25540)
==> Writing new distribution file: melodic/distribution.yaml
==> git add melodic/distribution.yaml
==> git commit -m "jderobot_color_tuner: 0.0.5-2 in 'melodic/distribution.yaml' [bloom]"
[bloom-jderobot_color_tuner-2 6e00f1638] jderobot_color_tuner: 0.0.5-2 in 'melodic/distribution.yaml' [bloom]
 1 file changed, 1 insertion(+), 1 deletion(-)
==> Pushing changes to fork
Counting objects: 4, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (3/3), done.
Writing objects: 100% (4/4), 419 bytes | 419.00 KiB/s, done.
Total 4 (delta 2), reused 0 (delta 0)
remote: Resolving deltas: 100% (2/2), completed with 2 local objects.
remote: 
remote: Create a pull request for 'bloom-jderobot_color_tuner-2' on GitHub by visiting:
remote:      https://github.com/pankhurivanjani/rosdistro/pull/new/bloom-jderobot_color_tuner-2
remote: 
To https://github.com/pankhurivanjani/rosdistro.git
 * [new branch]          bloom-jderobot_color_tuner-2 -> bloom-jderobot_color_tuner-2
<== Pull request opened at: https://github.com/ros/rosdistro/pull/25543

```

clalancette, 2020-06-22:
I see at least 2 problems in here:

```
This version of bloom is '0.9.1', but the newest available version is '0.9.7'. Please update.
```

Also:

```
Upstream Devel Branch:
  <vcs reference>
    Branch in upstream repository on which to search for the version.
    This is used only when version is set to ':{auto}'.
  [None]: 
```

You should:
1.  Update bloom to the latest version.  There are a lot of changes and bugfixes since 0.9.1.
1.  Provide an upstream development branch for bloom to use properly.

dirk-thomas, 2020-09-01:
@pankhurivanjani Friendly ping.

pankhurivanjani, 2020-09-02:
I am sorry for not getting back on the issue. This indeed solved the issues, I would close it. Thanks !

closed 2020-09-02 by pankhurivanjani