ros/rosdistro #25432: recommend / use vcstool instead of wstool

dirk-thomas, 2020-06-11:
[wstool](https://github.com/vcstools/wstool) as well as its dependency [vcstools](https://github.com/vcstools/vcstools) are not actively being maintained for quite some time (see the list of pending issues and pull requests).

With the move to Python 3 in ROS 1 Noetic the option was to:
* either make sure `wstool` is being updated to continue to work well which includes maintaining it for the supported lifetime of Noetic which is until April 2025,
* or switch to recommending `vcstool` instead which is actively being maintained and used for ROS 2 across all major platforms.

Therefore for ROS 1 Noetic the install instructions [[1](wiki.ros.org/noetic/Installation/Source)] recommend the use of `vcstool` to avoid having to spend any effort on maintaining a parallel set of packages for the same task.

---

In order to teach new users the use of `vcstool` instead a few additional wiki pages have been updated:
* http://wiki.ros.org/vcstool
* http://wiki.ros.org/action/diff/wstool?action=diff&rev1=15&rev2=16
* http://wiki.ros.org/action/diff/rosinstall_generator?action=diff&rev1=12&rev2=13
* http://wiki.ros.org/action/diff/melodic/Installation/Source?action=diff&rev1=7&rev2=8
* http://wiki.ros.org/action/diff/Debian/Installation/Source?action=diff&rev1=13&rev2=14
* http://wiki.ros.org/action/diff/catkin/Tutorials/workspace_overlaying?action=diff&rev1=33&rev2=34

---

That being said the use of `wstool` is documented in a lot of other places and neither the Debian package nor the PyPI package for `wstool` (and `vcstools`) will go away. So users can continue to be use them as before.

dirk-thomas, 2020-06-11:
The only piece of ROS infrastructure which was relying on `wstool` until now was prerelease.ros.org. As of ros-infrastructure/prerelease_website#61 it has been updated to use `vcstool`.

dirk-thomas, 2020-06-11:
To clearly communicate this also to user which are looking at the `vcstools` org unit directly I have created the above referenced tickets proposing to archive the repos.

dirk-thomas, 2020-06-25:
All open tickets in these repositories have been closed and the repositories have been archived.

closed 2020-06-25 by dirk-thomas

mikepurvis, 2020-06-29:
FYI @tspicer01