ros/rosdistro #25457: Ros noetic Ubuntu 20 Arm64 Raspberry pi 4 - unable to find package desktop full

Chick92, 2020-06-13:
I'm having trouble installing ros desktop full on my pi4, running ubuntu 20 64 bit, has it actually been released, or is it just listed as released because the X86 variant has been?

I can't install ubuntu 20 as my pi4 is the new 8gb RAM version, and the firmware on the ub18 iso hasn't been updated yet - it won't boot.

clalancette, 2020-06-14:
Sorry, this is a confusing report.  Are you using Ubuntu 18.04 or 20.04 for your Raspberry Pi?

We only support Ubuntu 20.04 binary packages for Noetic.  So if you are trying to install it on 18.04, it just won't work, it isn't a supported platform.  See https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025 for the supported platforms for Noetic.

nuclearsandwich, 2020-06-15:
According to the [ROS noetic Ubuntu Focal armhf status page](http://repositories.ros.org/status_page/ros_noetic_ufhf.html?q=desktop_full) the desktop_full package has not yet been built. Likely due to unavailable dependencies for that platform.

dirk-thomas, 2020-09-01:
@sloretz To determine what is missing and if that hole can be filled or not.

sloretz, 2020-09-11:
Cause is this failure to build gazebo_ros on Ubuntu Focal armhf

http://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__gazebo_ros__ubuntu_focal_armhf__binary/122/

```
19:37:28 The following packages have unmet dependencies:
19:37:28  ros-noetic-gazebo-dev : Depends: gazebo11 but it is not installable
19:37:28                          Depends: libgazebo11-dev but it is not installable
19:37:29 E: Unable to correct problems, you have held broken packages.
```

@chapulina Is Gazebo 11 supported on armhf? I didn't [find a gazebo-install-gazebo11 job for it](https://build.osrfoundation.org/view/main/view/gazebo/search/?q=gazebo-install-gazebo11), but I'm not 100% sure that's what the job would be called.

chapulina, 2020-09-11:
No armhf for Gazebo 11: http://packages.osrfoundation.org/gazebo/ubuntu-stable/pool/main/g/gazebo11/

Another related issue: https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1158

@j-rivero would know what's missing to release Gazebo 11 for armhf and how much effort that would be.

j-rivero, 2020-09-11:
> @j-rivero would know what's missing to release Gazebo 11 for armhf and how much effort that would be.

We don't any official support for armhf at this moment in Gazebo11 although it probably might be added without the need to patch software extensively.

One question @Chick92 : according to (raspberry pi4 model B page](https://www.raspberrypi.org/products/raspberry-pi-4-model-b/specifications/) the version of ARM is armv8 aka aarch64 aka arm64. Is this correct? If that is the case, then I would repeat my [this comment](https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1158#issuecomment-691229770).

sloretz, 2020-11-10:
It looks like Gazebo 11 isn't supported on `armhf`, but it is for `armv8`, and there is a `ros-desktop-full` package for that. http://repositories.ros.org/status_page/ros_noetic_ufv8.html?q=desktop_full . Based on @j-rivero's comment an `armv8` image should work on a Raspberry Pi 4.

@Chick92 What's the error you get when trying to install `ros-desktop-full` on `Arm64`?

closed 2021-11-08 by clalancette