ros/rosdistro #25516: replacing master/slave with an alternative

trusktr, 2020-06-18:
It may be more respectful for all possible users if the master/slave terminology is replaced with something else, like primary/assistant or something.

tfoote, 2020-06-19:
We've already made an effort to avoid using this terminology in several places in the ROS 1 codebase over the last several years. This particular instance is baked in deeply but is something that we have explicitly designed out of ROS 2. As such since ROS 1's final release, noetic, has already come out, this entire API will not be included in future ROS distributions. As such I would suggest that we consider this resolved since the terminology for this API will not be in any future rosdistro.

dirk-thomas, 2020-09-01:
Assuming the previous comment answered the question I will go ahead and close this ticket for now. Please feel free to continue commenting.

closed 2020-09-01 by dirk-thomas