ros/rosdistro #25760: What is the recommendation for adding Ignition Robotics Libraries?

Levi-Armstrong, 2020-07-10:
Ignition robotics provides several libraries along with several versions for each.  Example is show below for ignition-common. Should each of these be added to the base.yml file as separate entries?

```
libignition-common
libignition-common2
libignition-common2-dev
libignition-common3
libignition-common3-av
libignition-common3-av-dev
libignition-common3-core-dev
libignition-common3-dev
libignition-common3-events
libignition-common3-events-dev
libignition-common3-graphics
libignition-common3-graphics-dev
libignition-common3-profiler
libignition-common3-profiler-dev
libignition-common-dev
```

Levi-Armstrong, 2020-07-22:
Should I go through and add all of these?

tfoote, 2020-07-22:
@chapulina Just added several in #25789  We don't have the full granularity of the sub libraries. Do you need those specific sublibraries?

chapulina, 2020-07-23:
As @tfoote mentioned, we recently added `rosdep` support for Ignition libraries. These are the current limitations:

* Only the [Citadel versions of each library](https://ignitionrobotics.org/docs/citadel) are available (from your example, only `.*common3.*`)
* They're only available for Focal and Buster (i.e. Noetic and Foxy)

At a first pass, I didn't want to add every component unless it was necessary. That's why we currently only have the following 2 keys for Ignition Common:

* `ignition-common3`: Installs `libignition-common3-dev`, which itself pulls all other `libignition-common3-*-dev`. This key also works with `colcon` if you're building Ignition and ROS packages from source in the same workspace. I recommend you always use this one.
* `libignition-common3`: Installs `libignition-common3`, which itself pulls all other `libignition-common3-*`. This can be useful if you're really only interested in the runtime libraries.

The same goes for all other Ignition libraries, with the exception of Ignition Math, which has `ignition-math6`, `libignition-math6` *and* `ignition-math6-eigen3`. That was added because none of the former 2 will pull the latter.

Levi-Armstrong, 2020-07-23:
This is the resolution I currently need and thank you for the details on the the different keys available. This has been addressed by PR #25789.

closed 2020-07-23 by Levi-Armstrong