ros/rosdistro #26438: {Proposal} Adding fine-grained PCL dependencies

YoshuaNava, 2020-09-03:
Dear Developers,
My name is Yoshua Nava. I'm a Perception Software Engineer at ANYbotics AG (Zurich, Switzerland)

I'm writing this issue in relation to https://github.com/ros-perception/perception_pcl/issues/307. As mentioned in the perception_pcl issue, we would like to have more fine-grained PCL dependencies in the perception_pcl package, by adding the corresponding Debians for pcl modules to rosdistro.

Some of the packages that could be added (although I need to confirm this), are:

```
libpcl-common
libpcl-conversions-dev
libpcl-features
libpcl-filters
libpcl-io
libpcl-kdtree
libpcl-keypoints
libpcl-ml
libpcl-msgs-dev
libpcl-octree
libpcl-outofcore
```

I would like to propose adding the corresponding rosdep keys to the base.yaml file. Would the maintainers approve of this initiative? If affirmative I can open the corresponding pull request.

Thank you in advance.

Best regards,
Yoshua Nava

dirk-thomas, 2020-09-03:
> Would the maintainers approve of this initiative? If affirmative I can open the corresponding pull request.

Absolutely, please go ahead and create a pull request.

YoshuaNava, 2020-09-03:
@dirk-thomas Thank you very much for your prompt reply.

One question: I'm currently preparing a pull request. For Ubuntu and Debian it is straightforward, as the name of the pcl modules is quite similar. But for the rest of OS (arch, fedora, freebsd, gentoo, macports, openembedded, opensuse, rhel and slackware) I'm unsure about the way we should go about them.

Where could I find the name of those individual modules for the mentioned OS? (if they exist)

Thank you.

Best,
Yoshua Nava

dirk-thomas, 2020-09-03:
The [CONTRIBUTING](https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#expected-rosdep-sources) file contains some references how to search for available packages e.g. for Fedora.

YoshuaNava, 2020-09-07:
Sorry, had to pause working on this over the weekend. Resuming now.

closed 2020-11-12 by YoshuaNava