ros/rosdistro #26454: nlohmann-json-dev entry on Ubuntu 20.04 broken

dseifert, 2020-09-03:
#24432 adds nlohmann-json-dev to rosdistro and resolves it for all Ubuntu distros (except xenial) to nlohmann-json-dev. However, this package does not exist at least since Ubuntu 19.10:

- https://packages.ubuntu.com/search?keywords=nlohmann-json-dev only lists bionic
- https://packages.ubuntu.com/search?keywords=nlohmann-json3-dev shows eoan and newer

Note that nlohmann-json-dev installs nlohmann-json 2.x, and nlohmann-json3-dev installs 3.x.

Possible solutions:
- nlohmann-json-dev installs the latest version (2.x on bionic, 3.x on focal or newer)
- limit nlohmann-json-dev to bionic and add nlohmann-json3-dev definition for focal and newer)

Before creating a pull request, I wanted to collect input what the preferable solution would be (or whether I am somehow mistaken).

dseifert, 2020-09-03:
Actually the same could be true for Debian: https://packages.debian.org/search?keywords=nlohmann-json&searchon=names&suite=all&section=all where nlohmann-json-dev is not available in the bullseye testing release (or in sid), but nlohmann-json3-dev is

hidmic, 2020-09-07:
Hmm, as far as I can see, you're right @dseifert. `nlohmann-json-dev` is gone in `focal`. 

I'd be inclined towards solution (1.). Although https://github.com/nlohmann/json/issues/1938 seems to suggest the intent (at in Debain) was to have both `nlohmann-json-dev` *and* `nlohmann-json3-dev` released, the former has since been dropped. So I don't see the need for having two keys (TBH I don't see why the debian got renamed either). @dirk-thomas WDYT?

dirk-thomas, 2020-09-10:
One key sounds fine to me. The wildcard should match future distros and existing distros with a different value should be enumerated explicitly.

PR #26642 (2020-09-22): Implement fix to nlohmann-json-dev
As proposed in issues #26454, change the newer Ubuntu variants to use `nlohmann-json3-dev`.

* Ubuntu [Focal, Groovy]: https://packages.ubuntu.com/search?keywords=nlohmann-json3-dev
* Ubuntu [Bionic]: https://packages.ubuntu.com/search?keywords=nlohmann-json-dev
* Debian [Stretch]: https://packages.debian.org/search?keywords=nlohmann-json&searchon=names&suite=all&section=all
* Debian [ Buster, Bullseye, Sid]: https://packages.debian.org/search?keywords=nlohmann-json3&searchon=names&suite=all&section=all

Closes #26454.
commit a11e9afd (2020-09-22): Implement fix to nlohmann-json-dev (#26642)

* Implement fix to nlohmann-json-dev

Closes #26454.
closed 2020-09-22 by mjcarroll