ros/rosdistro #26526: Build issues on Debian Buster for Noetic release

rhaschke, 2020-09-11:
As originally pointed out in https://github.com/ros/rosdistro/pull/26484#issuecomment-688687613 we have a fundamental [build issue on Debian Buster](http://build.ros.org/job/Ndev_db__moveit__debian_buster_amd64/1), blocking the MoveIt release for Noetic. Due to some [change introduced since gcc 5](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70129) unnormalized or symlinked `-isystem` include paths that actually point to `/usr/include` are not handled correctly and screw up the include order. 
The issue was [finally resolved in cmake 3.16](https://gitlab.kitware.com/cmake/cmake/-/issues/16291#related-merge-requests). (Debian Buster vs. Ubuntu Focal use cmake version 3.13 vs. 3.16, which is why the issue only shows up in Buster.)

As suggested in https://github.com/ros/rosdistro/pull/26484#issuecomment-690394914, the issue should be resolved by normalizing exported include directories in all upstream packages. One offending package is `orocos-kdl`: `/usr/share/orocos_kdl/cmake/../../../include`. As this is a Debian system package, it might not be easy to get a change released there soon. @jspricke, I have seen you are a maintainer of this package. Can can you comment on this please?

jspricke, 2020-09-11:
Hi Robert,

I've just uploaded version 1.4.0-11 with a fix to unstable, should be 
available later today. Do you have some simple way to test the fix 
(simply edit the file as root ;) ) to verify?

I will submit it for the next point release later today.

Cheers Jochen

rhaschke, 2020-09-11:
I can confirm that using `/usr/include` instead of `/usr/share/orocos_kdl/cmake/../../../include` resolves the build problem.
Note, @sloretz, that many ROS packages need to be rebuild after this release as well, because they directly or indirectly depend on orocos-kdl and have the wrong include path stored in their cmake config (in /opt/ros/noetic/share) as well.
@jspricke, how fast do you think this fixup will show up in the official package repos? Should we think about an intermediate workaround anyway?

jspricke, 2020-09-11:
* Robert Haschke <notifications@github.com> [2020-09-11 01:23]:
>I can confirm that using `/usr/include` instead of `/usr/share/orocos_kdl/cmake/../../../include` resolves the build problem.

Do you have a minimal example to show the broken behavior?
It would simplify the discussion.

>@jspricke, how fast do you think this fixup will show up in the official package repos?

see: https://lists.debian.org/debian-release/2020/09/msg00072.html

rhaschke, 2020-09-11:
@sloretz: I can confirm that `orocos_kdl` is the only offending package. A local build in a Buster docker container succeeded.

dirk-thomas, 2020-09-11:
> I can confirm that `orocos_kdl` is the only offending package. A local build in a Buster docker container succeeded.

That is exactly the reason I didn't want to start with a global workaround in `catkin` but first identify which packages are actually causing this. I am glad it was only a single package which needed fixing.

jspricke, 2020-09-11:
This is tracked in https://bugs.debian.org/970098

rhaschke, 2020-09-14:
@jspricke, the minimal example is as simple as pointed out in the [original gcc bug report](https://gcc.gnu.org/bugzilla/show_bug.cgi?id=70129):
```shell
# echo "#include<algorithm>" | g++ -x c++ -isystem /usr/share/orocos_kdl/cmake/../../../include -
In file included from /usr/include/c++/8/bits/stl_algo.h:59,
                 from /usr/include/c++/8/algorithm:62,
                 from <stdin>:1:
/usr/include/c++/8/cstdlib:75:15: fatal error: stdlib.h: No such file or directory
 #include_next <stdlib.h>
               ^~~~~~~~~~
compilation terminated.
```
Any `-isystem` include path resolving to /usr/include will cause this error. Of course the best way to fix this issue (not only in orocos-kdl, but once for all) is to upgrade cmake to 3.16, e.g. from [buster-backports](https://packages.debian.org/buster-backports/cmake).

jspricke, 2020-09-19:
The update has been accepted and should be in buster on the 26.

rhaschke, 2020-09-29:
> The update has been accepted and should be in buster on the 26.

@jspricke, did you hear anything regarding the Buster release? The new version isn't yet available in Buster:
https://packages.debian.org/source/buster/orocos-kdl

jspricke, 2020-09-29:
Am 29. September 2020 20:55:41 MESZ schrieb Robert Haschke <notifications@github.com>:
>> The update has been accepted and should be in buster on the 26.
>
>@jspricke, did you hear anything regarding the Buster release? The new version isn't yet available in Buster:
>https://packages.debian.org/source/buster/orocos-kdl

This is the new version, deb10u2.
See the changelog: https://metadata.ftp-master.debian.org/changelogs//main/o/orocos-kdl/orocos-kdl_1.4.0-7+deb10u2_changelog

rhaschke, 2020-09-29:
Ah. Ok. I was expecting 1.4.0-11 as indicated by the source repo: https://salsa.debian.org/science-team/orocos-kdl.

jspricke, 2020-09-29:
Am 29. September 2020 21:09:13 MESZ schrieb Robert Haschke <notifications@github.com>:
>Ah. Ok. I was expecting 1.4.0-11 as indicated by the source repo: https://salsa.debian.org/science-team/orocos-kdl.

That's the version in testing, stable updates are only targeted fixes, see the bug linked above as well.

rhaschke, 2020-09-30:
The binary build [still fails on Buster](http://build.ros.org/view/Nsrc_dB/job/Nbin_db_dB64__moveit_core__debian_buster_amd64__binary/).
Although the most recent `liborocos-kdl-dev: 1.4.0-7+deb10u2` is installed, the include path didn't yet changed:
`-isystem /usr/share/orocos_kdl/cmake/../../../include`

@sloretz: It looks like that some upstream ROS packages still have this old orocos include path in their cmake config, e.g.:
```
/opt/ros/noetic/share/tf2_kdl/cmake/tf2_kdlConfig.cmake:  set(_include_dirs "include;/usr/include/eigen3;/usr/share/orocos_kdl/cmake/../../../include")
```
I guess all ROS packages depending on liborocos-kdl-dev should be rebuild:
```sh
# apt rdepends liborocos-kdl-dev
liborocos-kdl-dev
Reverse Depends:
  Depends: ros-noetic-kdl-parser (>= 1.3.0)
  Depends: ros-noetic-tf2-kdl
  Depends: ros-noetic-tf2-geometry-msgs
  Depends: ros-noetic-tf-conversions
  Depends: ros-noetic-robot-state-publisher
  Recommends: science-robotics-dev
  Depends: ros-noetic-kdl-conversions
  Depends: ros-noetic-eigen-conversions
  Depends: librobot-state-publisher-dev
  Depends: ros-robot-dev
  Depends: libkdl-parser-dev
  Depends: libtf2-kdl-dev
  Depends: libtf2-geometry-msgs-dev
  Depends: libtf-conversions-dev
```

sloretz, 2020-09-30:
> I guess all ROS packages depending on liborocos-kdl-dev should be rebuild:

Almost every package should have rebuilt with https://github.com/ros/rosdistro/pull/26742 and https://github.com/ros/rosdistro/pull/26744 . Using the ros-testing repo on buster I no longer see the orocos_kdl path in tf2_kdl

```CMake
   set(_include_dirs "include;/usr/include/eigen3")   
```

sloretz, 2020-09-30:
Looks like `ros-noetic-eigen-conversions` and `ros-noetic-kdl-conversions` still reference the old paths

```
apt install ros-noetic-kdl-parser ros-noetic-tf2-kdl ros-noetic-tf2-geometry-msgs ros-noetic-tf-conversions ros-noetic-robot-state-publisher ros-noetic-kdl-conversions ros-noetic-eigen-conversions
```

```console
$ grep /opt/ros/noetic/ -R -e "orocos_kdl/cmake/"
/opt/ros/noetic/lib/pkgconfig/eigen_conversions.pc:Cflags: -I${prefix}/include -I/usr/include/eigen3 -I/usr/share/orocos_kdl/cmake/../../../include
/opt/ros/noetic/lib/pkgconfig/kdl_conversions.pc:Cflags: -I${prefix}/include -I/usr/share/orocos_kdl/cmake/../../../include -I/usr/include/eigen3
/opt/ros/noetic/share/kdl_conversions/cmake/kdl_conversionsConfig.cmake:if(NOT "include;/usr/share/orocos_kdl/cmake/../../../include;/usr/include/eigen3 " STREQUAL " ")
/opt/ros/noetic/share/kdl_conversions/cmake/kdl_conversionsConfig.cmake:  set(_include_dirs "include;/usr/share/orocos_kdl/cmake/../../../include;/usr/include/eigen3")
/opt/ros/noetic/share/eigen_conversions/cmake/eigen_conversionsConfig.cmake:if(NOT "include;/usr/include/eigen3;/usr/share/orocos_kdl/cmake/../../../include " STREQUAL " ")
/opt/ros/noetic/share/eigen_conversions/cmake/eigen_conversionsConfig.cmake:  set(_include_dirs "include;/usr/include/eigen3;/usr/share/orocos_kdl/cmake/../../../include")
```

sloretz, 2020-09-30:
Rebuilding `eigen_conversions` and `kdl_conversions` seems to have gotten rid of the last references to these paths. If I understand correctly, I think this issue will be resolved by the next Noetic sync.

```
Setting up ros-noetic-eigen-conversions (1.13.2-1buster.20200930.155358) ...
Setting up ros-noetic-kdl-conversions (1.13.2-1buster.20200930.155542) ...
Setting up ros-noetic-tf-conversions (1.13.2-1buster.20200930.161300) ...
```

```console
$ grep /opt/ros/noetic/ -R -e "orocos_kdl/cmake/"
$
```

rhaschke, 2020-09-30:
As the ROS buildfarm successfully built MoveIt, I'm closing this. Thanks for all your contributions!

closed 2020-09-30 by rhaschke