ros/rosdistro #26853: Invalid branches / repos

dirk-thomas, 2020-10-07:
**Kinetic**

* [ ] https://github.com/DroidAITech/xbot `master` #26850
* [ ] https://github.com/RethinkRobotics-opensource/sns_ik `master`
* [ ] https://github.com/RoverRobotics/rr_openrover_basic
* [ ] https://github.com/Terabee/teraranger_array `ros-release`
* [ ] https://github.com/allenh1/p2os `master`
* [ ] https://github.com/dartsim/dart `release-6.3`
* [ ] https://github.com/durovsky/binpicking_utils
* [ ] https://github.com/hansonrobotics/hr_msgs
* [ ] https://github.com/injones/mycroft_ros `kinetic-devel`
* [ ] https://github.com/ros-simulation/stage_ros `master`
* [ ] https://github.com/roverrobotics/rr_swiftnav_piksi
* [ ] https://github.com/stonier/py_trees_msgs `devel`
* [ ] https://github.com/stonier/sophus `indigo`
* [ ] https://github.com/yujinrobot/yocs_msgs `kinetic`

**Melodic**

* [ ] https://github.com/ROBOTIS-GIT/open_manipulator_with_tb3
* [ ] https://github.com/allenh1/p2os `master`
* [ ] https://github.com/ethz-asl/rotors_simulator `feature/gazebo9-autobackport`
* [ ] https://github.com/stonier/py_trees_msgs `release/0.3-melodic`

**Noetic**

* [ ] https://github.com/allenh1/p2os `master`

**Dashing**

* [ ] https://github.com/boschresearch/fmi_adapter_ros2 `dashing`
* [ ] https://github.com/ros-security/launch_ros_sandbox `dashing`

**Eloquent**

* [ ] https://github.com/boschresearch/fmi_adapter_ros2 `eloquent`

hidmic, 2020-10-14:
@dirk-thomas shall we try to ping the repo owners?

dirk-thomas, 2020-10-14:
> shall we try to ping the repo owners?

Each ROS boss should either do that or just update the entries if that there is an obviously correct value. (The latter seems to more efficient to me.)

dirk-thomas, 2020-10-20:
Thanks @sloretz and @clalancette for checking off the items for Noetic and Melodic.

@mjcarroll @nuclearsandwich @tfoote Friendly ping.

dirk-thomas, 2020-10-20:
@nuclearsandwich Btw. there is no `dashing` label atm.

mjcarroll, 2020-11-10:
Friendly ping @tfoote, as most remaining are `kinetic`.

tfoote, 2021-04-24:
@mxgrey  Could you look at what the appropriate branch is for Dartsim?

mxgrey, 2021-04-24:
If we're allowed to use tag names rather than branches, then tag `v6.3.1` should be the right replacement for dartsim.

mxgrey, 2021-04-24:
@tfoote I've opened a PR, assuming tags are okay: https://github.com/ros/rosdistro/pull/29279

mxgrey, 2021-04-24:
I've reopened the `release-6.3` branch for dartsim, so hopefully it should start working again.

pl-kabaradjian, 2021-04-26:
hello @tfoote
I can confirm that **ros-release** is the intended branch (not updated I admit). Was there an issue with it?

mxgrey, 2021-04-26:
I got [this indication](https://build.ros.org/job/Kdev__dartsim__ubuntu_xenial_amd64/2/) that `dartsim` is okay again, so I think it can be checked off.

tfoote, 2021-05-08:
I opened a PR for the last dashing error flagged #29461 and there's only one kinetic one left for a non-extant repo that's long ticketed upstream. I'm going to close this out.