ros/rosdistro #27023: there is an issue with this and boost system linkage

brookdpg, 2020-10-19:
ops@ops-XPS-15-9500:~/ws_ros2$ colcon build --event-handlers desktop_notification- status- --cmake-args -DCMAKE_BUILD_TYPE=Release
Starting >>> geometric_shapes
Starting >>> moveit_msgs
Starting >>> srdfdom
Starting >>> moveit_common
Starting >>> hardware_interface
Starting >>> moveit_resources_panda_description
Starting >>> moveit_resources_fanuc_description
Starting >>> moveit_resources_pr2_description
Finished <<< moveit_resources_panda_description [0.21s]
Finished <<< moveit_resources_fanuc_description [0.22s]
Starting >>> moveit_resources_panda_moveit_config
Starting >>> moveit_resources_fanuc_moveit_config
Finished <<< moveit_common [0.25s]
Finished <<< moveit_resources_pr2_description [0.26s]
Finished <<< srdfdom [0.31s]
Finished <<< moveit_resources_panda_moveit_config [0.18s]
Finished <<< hardware_interface [0.43s]
Finished <<< moveit_resources_fanuc_moveit_config [0.20s]
Starting >>> controller_interface
Starting >>> test_robot_hardware
Starting >>> moveit_resources
Starting >>> transmission_interface
Finished <<< moveit_resources [0.12s]
Finished <<< controller_interface [0.17s]
Finished <<< test_robot_hardware [0.18s]
Starting >>> controller_manager
--- stderr: geometric_shapes
/usr/bin/ld: CMakeFiles/test_body_operations.dir/test_body_operations.cpp.o: in function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
test_body_operations.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x44): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_body_operations.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xc1): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_body_operations.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x151): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: CMakeFiles/test_body_operations.dir/test_body_operations.cpp.o: in function `boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const':
test_body_operations.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0x43): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_body_operations.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0xd1): undefined reference to `boost::system::generic_category()'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_body_operations.dir/build.make:180: test/test_body_operations] Error 1
make[1]: *** [CMakeFiles/Makefile2:499: test/CMakeFiles/test_body_operations.dir/all] Error 2
/usr/bin/ld: CMakeFilesmake[1]: *** Waiting for unfinished jobs....
/test_point_inclusion.dir/test_point_inclusion.cpp.o: in function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
test_point_inclusion.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x44): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_point_inclusion.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xc1): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_point_inclusion.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x151): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: CMakeFiles/test_point_inclusion.dir/test_point_inclusion.cpp.o: in function `boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const':
test_point_inclusion.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0x43): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_point_inclusion.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0xd1): undefined reference to `boost::system::generic_category()'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_point_inclusion.dir/build.make:180: test/test_point_inclusion] Error 1
make[1]: *** [CMakeFiles/Makefile2:354: test/CMakeFiles/test_point_inclusion.dir/all] Error 2
/usr/bin/ld: /home/ops/ros2_foxy/install/random_numbers/lib/librandom_numbers.so: undefined reference to `boost::system::generic_category()'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_bounding_box.dir/build.make:180: test/test_bounding_box] Error 1
make[1]: *** [CMakeFiles/Makefile2:296: test/CMakeFiles/test_bounding_box.dir/all] Error 2
/usr/bin/ld: CMakeFiles/test_loaded_meshes.dir/test_loaded_meshes.cpp.o: in function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
test_loaded_meshes.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x44): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_loaded_meshes.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xc1): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_loaded_meshes.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x151): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: CMakeFiles/test_loaded_meshes.dir/test_loaded_meshes.cpp.o: in function `boost::system::error_category::std_category::equivalent(int, std::error_condition const&) const':
test_loaded_meshes.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0x43): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_loaded_meshes.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition[_ZNK5boost6system14error_category12std_category10equivalentEiRKSt15error_condition]+0xd1): undefined reference to `boost::system::generic_category()'
collect2: error: ld returned 1 exit status
make[2]: *** [test/CMakeFiles/test_loaded_meshes.dir/build.make:180: test/test_loaded_meshes] Error 1
make[1]: *** [CMakeFiles/Makefile2:412: test/CMakeFiles/test_loaded_meshes.dir/all] Error 2
/usr/bin/ld: CMakeFiles/test_bounding_sphere.dir/test_bounding_sphere.cpp.o: in function `boost::system::error_category::std_category::equivalent(std::error_code const&, int) const':
test_bounding_sphere.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x44): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_bounding_sphere.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0xc1): undefined reference to `boost::system::generic_category()'
/usr/bin/ld: test_bounding_sphere.cpp:(.text._ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei[_ZNK5boost6system14error_category12std_category10equivalentERKSt10error_codei]+0x151): undefined reference to `boost::system::generic_category()'
[... cut ...]
make[1]: *** [CMakeFiles/Makefile2:239: test/CMakeFiles/test_ray_intersection.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
---
Failed   <<< geometric_shapes [0.66s, exited with code 2]
Aborted  <<< controller_manager [0.01s]
Aborted  <<< transmission_interface [0.18s]
Aborted  <<< moveit_msgs [1.13s]

Summary: 11 packages finished [1.44s]
  1 package failed: geometric_shapes
  3 packages aborted: controller_manager moveit_msgs transmission_interface
  1 package had stderr output: geometric_shapes
  21 packages not processed
ops@ops-XPS-15-9500:~/ws_ros2$

hidmic, 2020-10-20:
@brookdpg would you mind providing us with a bit more context? You may check the [ROS 2 Developer Guide](https://index.ros.org/doc/ros2/Contributing/Developer-Guide/#issues) for some basic guidelines.

dirk-thomas, 2020-10-26:
@brookdpg Friendly ping.

mjcarroll, 2020-11-10:
@brookdpg without more information, I don't think that this issue can be resolved.  I'm closing this due to lack of activity.

closed 2020-11-10 by mjcarroll