ros/rosdistro #27060: normal wait time for recently merged packages to become available?

flynneva, 2020-10-21:
i recently merged two new interface packages into the ros2 branches. for one of the packages it was merged in 9 days ago and for the other 7 days ago.

both packages are not yet available to apt-get install yet. I was expecting some delay (~5 days) before they became available but now i am starting to get worried.

these packages are udp_msgs and grbl_msgs for reference.

any ideas on when they might become available to install?

dirk-thomas, 2020-10-21:
What is the upstream repository? Did you create a new release? Did you run `bloom-release`? Can you reference the ros/rosdistro PR where that version bump was made?

flynneva, 2020-10-21:
well I made a release for each ROS2 distro, and all were merged in.

you can see them just by [filtering the closed PR's by my user name](https://github.com/ros/rosdistro/pulls?q=is%3Apr+is%3Aclosed+flynneva)

flynneva, 2020-10-21:
i guess what im more looking for though is just a generic timeline....like is there a specific day of the week that updates are pushed to the public for the apt deb packages? or is it just random?

dirk-thomas, 2020-10-21:
Shortly (depending on the buildfarm load) after the `ros/rosdistro` PR is merged the Debian packages will be built, e.g. in http://build.ros2.org/view/Fbin_uF64/job/Fbin_uF64__grbl_msgs__ubuntu_focal_amd64__binary/.

When the buildfarm has finished building Debian packages (including other packages as well as downstream packages of them) the set of updated Debian packages will be synced to the [testing](http://repositories.ros.org/ubuntu/testing/) repo (if no major regression has happened). So usually within a few hours your packages should appear there. You could use the testing repo (if you are willing to do so) or just pull your `deb`s from there to try them out.

The sync from `testing` to the `main` repo happens manually and is being announced on Discourse (e.g. https://discourse.ros.org/t/preparing-for-foxy-sync-and-patch-release-2020-10-09/16696) and usually happens every two weeks. Since the last sync was on Oct 11th and your package was built on Oct 14th it should be synced to main the next time - likely end of Oct.

closed 2020-10-21 by dirk-thomas