ros/rosdistro #27139: Release of "proj" into dashing broke swri_transform_util

pjreed, 2020-10-26:
The package [swri_transform_util](https://github.com/swri-robotics/marti_common/tree/dashing-devel/swri_transform_util) depends on the rosdep key `proj`, which, just for reference, maps to [libproj-dev in Ubuntu](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L5396).

Commit ac735e000956a3c3ab75725b5cd119f7f0f2dae8 released a package named "proj" into the ROS Dashing distribution, but it is currently [failing to build on the build farm](http://build.ros2.org/job/Dbin_uB64__proj__ubuntu_bionic_amd64__binary/); that in turn is causing swri_transform_util and everything that depends on it to fail because they are now trying to use that package to resolve their dependency.

What's the best way to go about fixing this?

dirk-thomas, 2020-10-26:
@nuclearsandwich @stonier FYI

I would suggest to revert #27049. It would probably be good if CI would catch this kind of collision.

dirk-thomas, 2020-10-26:
(@nuclearsandwich Btw. there is no Dashing label atm.)

dirk-thomas, 2020-10-27:
This should be resolved by #27142.

Please comment if there is still a problem.

closed 2020-10-27 by dirk-thomas