ros/rosdistro #27302: Today's release conflicts with npm on Ubuntu 18.04

brettle, 2020-11-11:
Installing ROS packages from http://packages.ros.org/ros/ubuntu/ on Ubuntu 18.04 results in npm being uninstalled, and installing npm causes the ROS packages to be uninstalled. This appears to be repeat of the behavior reported here:

https://github.com/ros/rosdistro/issues/19845#issuecomment-455641293

I'm not positive, but I *think* this just started today which is suspiciously the same day that the ros packages were updated.

clalancette, 2020-11-11:
Unfortunately, I don't know if there is anything we can do here.  Fundamentally, `libssl1.0-dev` is incompatible with `libssl-dev`.  npm depends on the former, and all of the ROS packages depend on the latter.

In point of fact, this should have always failed on 18.04.  Did this used to work with the npm packages from 18.04?

brettle, 2020-11-11:
Per my original message, I'm not sure if the issue is new. Is there an easy way I can install the ros packages from before today's release?

nuclearsandwich, 2020-11-11:
> Is there an easy way I can install the ros packages from before today's release?

You can use the [snapshot repositories](http://wiki.ros.org/SnapshotRepository): 
http://snapshots.ros.org/melodic/2020-10-12/

brettle, 2020-11-11:
Thanks! I can confirm that the problem is *not* new. I've changed the title to reflect of this issue to reflect that.
Sorry for the false alarm.

nuclearsandwich, 2020-11-12:
As mentioned in this comment https://github.com/ros/rosdistro/issues/19845#issuecomment-455737976 the nodejs version in bionic-security seems to link a different libssl version than the one in bionic-updates. Are you able to pin to that version of the nodejs package and keep everything installed?

brettle, 2020-11-13:
Yes, I can confirm that doing:

```
apt-get install nodejs=8.10.0~dfsg-2ubuntu0.2 nodejs-dev=8.10.0~dfsg-2ubuntu0.2
```

prevents a subsequent install of ros from removing npm.

tfoote, 2021-03-26:
I think that there's not much else that we can do except tell people to install the version of nodejs from bionic-security. Hopefully this ticket will be enough documentation for that. I'm going to close this as there's nothing actionable that we should be tracking.

closed 2021-03-26 by tfoote