ros/rosdistro #27359: How to properly handle test packages

kyrofa, 2020-11-17:
This is a bit of a meta-bug that can be solved in a number of ways across a few different repositories (including this one), so I'm proceeding with logging it here knowing that it may end up actually being solved elsewhere.

A given source repository in ROS can (and often does) contain multiple ROS packages. Each ROS package in that repo must explicitly be a part of the ROS distro by being contained in a release entry in the distribution.yaml, as well as being individually bloomed and ending up in the release repo.

However, not all packages within a given source repo MUST be part of the ROS distro. For example, take ros_comm, that has an [entire directory of test packages](https://github.com/ros/ros_comm/tree/noetic-devel/test). These test packages are [not contained in the distribution.yaml](https://github.com/ros/rosdistro/blob/master/noetic/distribution.yaml#L4488-L4508), nor are they bloomed, and thus they can't be resolved by rosdep into anything.

This makes sense, but it's also where problems can occur. These test packages, since they aren't in the ROS distro, seemingly are only useful for unit tests on the source repository itself. If they leak into the tests of packages that actually ARE part of the ROS distro, things become a little messy.

For example, take rosmsg, which is also part of ros_comm. It's a package that is actually part of the ROS distro, and is bloomed. However, it uses `test_rosmaster` [in its tests](https://github.com/ros/ros_comm/blob/6b1ee1a43b5b0a970d85612435e269818c89a152/tools/rosmsg/test/test_rosmsg_command_line.py#L78). `test_rosmaster` is one of ros_comm's test packages that is NOT part of the ROS distro. As a result, rosmsg doesn't (and indeed cannot) declare a dependency on it, but it uses it anyway. When rosmsg is fetched from the release repo (e.g. as part of rosinstall_generate, or the pre-release test in the build farm), its tests require a ROS package to be present that cannot possibly be present unless the entire ros_comm source repository is cloned.

That example can be fixed pretty easily by updating rosmsg to not have any implicit dependencies, and ensure it only requires things that can be satisfied by the ROS distro, but there might be better ways to solve this by exporting such test packages in the ROS distro in a way that they can be resolved and fetched properly without having to actually get binary test packages built. I wanted to start the discussion here. If it seems that fixing rosmsg is the way to go instead of a larger fix, I'm happy to make that PR.

nuclearsandwich, 2020-11-17:
Thanks for writing up the issue and linking it throughout. The only addition I have is that it isn't an accident these packages are not released they're being ignored by bloom explicitly since they provide no runtime utility to ROS end-user. https://github.com/ros-gbp/ros_comm-release/blob/master/noetic.ignored

Splitting this into two identified problems:
1. rosmsg would seem to have an undeclared test dependency
2. What expectations should we place on non-released packages when using rosinstall_generator?

I think the likely answer for 2 above is that released packages must not depend on unreleased ones and that correcting the rosmsg issue would eliminate the problem.

kyrofa, 2020-11-17:
> I think the likely answer for 2 above is that released packages must not depend on unreleased ones

I agree, although that does relegate test packages to only have meaning when running tests with the source repo, are we okay with that? I'll go ahead and propose a fix for rosmsg now.

nuclearsandwich, 2020-11-18:
> that does relegate (or rather, officially declare) test packages to only have meaning when running tests with the source repo, are we okay with that?

I think that this would just be a hardening (or re-hardening) of an existing de facto position. But I don't need to be the only voice from the ROS release team.

The buggy relationship between rosmsg and test_rosmaster aside, tests contained within packages can still be run from release checkouts and it would be possible to fetch test packages from source and run them against the release checkouts using an overlay workspace and some source ignores if the test package repository contains both test and non-test packages.

clalancette, 2020-11-30:
> > that does relegate (or rather, officially declare) test packages to only have meaning when running tests with the source repo, are we okay with that?
> 
> I think that this would just be a hardening (or re-hardening) of an existing de facto position.

I'm OK with that de facto position.  While I can imagine use cases for running integration/system tests on already released packages, presumably that was done *before* the release.  So it is not a strong use case in my opinion.  And if there *was* a strong use case, we could make an exception and release a test package.

tfoote, 2020-12-09:
If your package is designed to be reused by downstream packages it's not really a test package but a library  (which may be useful for testing) but in that case it should install the resources and be packaged and redistributed etc. And to that end those resources used for others tests probably want to be separated from your actual tests and either packaged into the package itself or isolated from the integration tests as the dependency count is likely much lower for the test fixtures or other reuseable components rather than the full integration tests.

Thus I'd advocate with sticking with the de facto standard that test packages aren't released. And if there are resources in a test package that someone requests be made available consider either moving those resources to a released package or actually releasing the test package on a case by case basis.

kyrofa, 2020-12-09:
Thank you all for your input. I've taken a crack at fixing rosmsg in https://github.com/ros/ros_comm/pull/2103 if you'd like to take a look.

kyrofa, 2020-12-09:
I would say this issue is resolved, I'll go ahead and close.

closed 2020-12-09 by kyrofa

nuclearsandwich, 2020-12-10:
Thanks for raising the issue and working on a fix!