ros/rosdistro #27401: ROS Noetic November 2020 Regressions

sloretz, 2020-11-20:
There are a lot of regressions in Noetic, and it's been a while since it had a sync. I think the following would get back to a sync'able state:

- [ ] Rollback moveit to `1.1.0-1` because that doesn't depend on `ros-noetic-fcl` (Related to #26527, means reverting #26938 @tylerjw FYI)
- [ ] Rollback ros_controllers to `0.17.0-1` because that version built (means reverting #26904 @bmagyar FYI)

## Ubuntu Focal amd64

## Ubuntu Focal arm64
* [chomp_motion_planner](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__chomp_motion_planner__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`
* [jsk_pcl_ros_utils](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__jsk_pcl_ros_utils__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_perception`
* [moveit](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_commander`, `moveit_core`, `moveit_planners`, `moveit_plugins`, `moveit_ros`, `moveit_setup_assistant`
* [moveit_chomp_optimizer_adapter](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_chomp_optimizer_adapter__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `chomp_motion_planner`, `moveit_core`
* [moveit_commander](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_commander__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_planning_interface`
* [moveit_core](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_core__ubuntu_focal_arm64__binary/)
  * :warning: Upstream packages that failed to build: `fcl`
* [moveit_fake_controller_manager](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_fake_controller_manager__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_planning`
* [moveit_kinematics](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_kinematics__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_planning`
* [moveit_planners](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_planners__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `chomp_motion_planner`, `moveit_planners_chomp`, `moveit_planners_ompl`
* [moveit_planners_chomp](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_planners_chomp__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `chomp_motion_planner`, `moveit_core`, `moveit_ros_planning_interface`
* [moveit_planners_ompl](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_planners_ompl__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_planning`
* [moveit_plugins](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_plugins__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_fake_controller_manager`, `moveit_ros_control_interface`, `moveit_simple_controller_manager`
* [moveit_ros](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_benchmarks`, `moveit_ros_manipulation`, `moveit_ros_move_group`, `moveit_ros_perception`, `moveit_ros_planning`, `moveit_ros_planning_interface`, `moveit_ros_robot_interaction`, `moveit_ros_visualization`, `moveit_ros_warehouse`
* [moveit_ros_benchmarks](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_benchmarks__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_planning`, `moveit_ros_warehouse`
* [moveit_ros_control_interface](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_control_interface__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_simple_controller_manager`
* [moveit_ros_manipulation](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_manipulation__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_move_group`, `moveit_ros_planning`
* [moveit_ros_move_group](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_move_group__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_kinematics`, `moveit_ros_planning`
* [moveit_ros_occupancy_map_monitor](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_occupancy_map_monitor__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`
* [moveit_ros_perception](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_perception__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_occupancy_map_monitor`
* [moveit_ros_planning](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_planning__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_occupancy_map_monitor`
* [moveit_ros_planning_interface](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_planning_interface__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_manipulation`, `moveit_ros_move_group`, `moveit_ros_planning`, `moveit_ros_warehouse`
* [moveit_ros_robot_interaction](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_robot_interaction__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_planning`
* [moveit_ros_visualization](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_visualization__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_perception`, `moveit_ros_planning_interface`, `moveit_ros_robot_interaction`, `moveit_ros_warehouse`
* [moveit_ros_warehouse](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_ros_warehouse__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_planning`
* [moveit_runtime](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_runtime__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_planners`, `moveit_plugins`, `moveit_ros_manipulation`, `moveit_ros_move_group`, `moveit_ros_perception`, `moveit_ros_planning`, `moveit_ros_planning_interface`, `moveit_ros_warehouse`
* [moveit_servo](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_servo__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_planning_interface`
* [moveit_setup_assistant](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_setup_assistant__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_planning`, `moveit_ros_visualization`
* [moveit_simple_controller_manager](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__moveit_simple_controller_manager__ubuntu_focal_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`

## Ubuntu Focal armhf
* [chomp_motion_planner](http://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__chomp_motion_planner__ubuntu_focal_armhf__binary/)
  * Upstream packages blocking this: `moveit_core`
* [jsk_pcl_ros_utils](http://build.ros.org/view/Nbin_ufhf_uFhf/job/Nbin_ufhf_uFhf__jsk_pcl_ros_utils__ubuntu_focal_armhf__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_perception`
[... 113 lines cut ...]
* [moveit_ros_occupancy_map_monitor](http://build.ros.org/view/Nbin_dbv8_dBv8/job/Nbin_dbv8_dBv8__moveit_ros_occupancy_map_monitor__debian_buster_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`
* [moveit_ros_perception](http://build.ros.org/view/Nbin_dbv8_dBv8/job/Nbin_dbv8_dBv8__moveit_ros_perception__debian_buster_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_occupancy_map_monitor`
* [moveit_ros_planning](http://build.ros.org/view/Nbin_dbv8_dBv8/job/Nbin_dbv8_dBv8__moveit_ros_planning__debian_buster_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_occupancy_map_monitor`
* [moveit_ros_planning_interface](http://build.ros.org/view/Nbin_dbv8_dBv8/job/Nbin_dbv8_dBv8__moveit_ros_planning_interface__debian_buster_arm64__binary/)
  * Upstream packages blocking this: `moveit_ros_manipulation`, `moveit_ros_move_group`, `moveit_ros_planning`, `moveit_ros_warehouse`
[... cut ...]
  * Upstream packages blocking this: `moveit_core`, `moveit_ros_planning`, `moveit_ros_visualization`
* [moveit_simple_controller_manager](http://build.ros.org/view/Nbin_dbv8_dBv8/job/Nbin_dbv8_dBv8__moveit_simple_controller_manager__debian_buster_arm64__binary/)
  * Upstream packages blocking this: `moveit_core`
* [ros_controllers](http://build.ros.org/view/Nbin_dbv8_dBv8/job/Nbin_dbv8_dBv8__ros_controllers__debian_buster_arm64__binary/)
  * :warning: Upstream packages that failed to build: `diff_drive_controller`
  * Upstream packages blocking this: `ackermann_steering_controller`

bmagyar, 2020-11-21:
I probably missed the announcement that added Debian builds to the requirements for a distro release. 
I'd be happy to run pre-release tests for debian too, could you please point me to a script generator for that?

tylerjw, 2020-11-21:
I can look into why fcl isn't building.  I was actually about to bloom moveit again yesterday (1.1.2) for Noetic but was stopped by a regression inside moveit.

If it makes sense to get a sync done in the short term, you can go ahead and back out what version moveit is at.  I didn't realize there was anything wrong with fcl.

I also have another package that keeps failing on buster that I can't seem to figure out.  I might just make a PR to remove it from rosdisyro for Noetic.

rhaschke, 2020-11-22:
Hi @sloretz, looking into the [failing FCL build for ARM architectures](http://build.ros.org/view/Nbin_ufv8_uFv8/job/Nbin_ufv8_uFv8__fcl__ubuntu_focal_arm64__binary/43/console#console-section-15), I noticed that your update to disable sse optimizations isn't yet applied: 
```
00:57:18 cd /tmp/binarydeb/ros-noetic-fcl-0.6.1/obj-aarch64-linux-gnu/src && /usr/lib/ccache/c++  -Dfcl_EXPORTS -I/tmp/binarydeb/ros-noetic-fcl-0.6.1/obj-aarch64-linux-gnu/include -I/tmp/binarydeb/ros-noetic-fcl-0.6.1/include -isystem /usr/include/eigen3 -isystem /opt/ros/noetic/include  -g -O2 -fdebug-prefix-map=/tmp/binarydeb/ros-noetic-fcl-0.6.1=. -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC   -std=c++11 -W -Wall -Wextra -Wpedantic -fPIC -fvisibility=hidden -mfpmath=sse -msse -msse2 -msse3 -mssse3 -o CMakeFiles/fcl.dir/broadphase/broadphase_SSaP.cpp.o -c /tmp/binarydeb/ros-noetic-fcl-0.6.1/src/broadphase/broadphase_SSaP.cpp
00:57:18 c++: error: unrecognized command line option ‘-mfpmath=sse’
00:57:18 c++: error: unrecognized command line option ‘-msse’
00:57:18 c++: error: unrecognized command line option ‘-msse2’
00:57:18 c++: error: unrecognized command line option ‘-msse3’
00:57:18 c++: error: unrecognized command line option ‘-mssse3’
```
I don't yet understand why. Did I make a mistake in the release, such that the release doesn't comprise your commits?

rhaschke, 2020-11-22:
I pulled the debian source package and indeed it doesn't comprise the latest release state.

sloretz, 2020-12-07:
> but subsequently, also an older tarball is downloaded: [...]  Shouldn't the debian source package built from the branch? What's the role of the tarball?

I'm not sure what the role of the older tarball is, (it seems to come from https://github.com/ros-infrastructure/ros_buildfarm/pull/374 ) but it looks like the output of the source job includes the updated code. One of the outputs of the job is an archive `/tmp/sourcedeb/ros-noetic-fcl_0.6.1-2focal.debian.tar.xz` which has a file `debian/patches/debian-changes-0.6.1-2focal` containing the changes between `0.6.1-2` and the original `0.6.1`.

sloretz, 2020-12-08:
> I probably missed the announcement that added Debian builds to the requirements for a distro release.

Oops, [Debian Buster is only Recommend Support for ROS Noetic](https://www.ros.org/reps/rep-0003.html#noetic-ninjemys-may-2020-may-2025). I chose to revert `ros_controllers` because `ros-noetic-desktop-full` depends on it, and it seems likely to be noticed [since that's the recommended metapackage to install](http://wiki.ros.org/noetic/Installation/Ubuntu#Installation-1).

> I'd be happy to run pre-release tests for debian too, could you please point me to a script generator for that?

I'd recommend installing the `python3-ros-buildfarm` package and [running the Debian Buster dev job locally](https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/devel_jobs.rst#run-the-devel-job-locally). For `ros_controllers` the commands to do the Debian Buster amd64 dev job locally appear to be

```
mkdir /tmp/devel_job
generate_devel_script.py https://raw.githubusercontent.com/ros-infrastructure/ros_buildfarm_config/production/index.yaml noetic db ros_controllers debian buster amd64 > /tmp/devel_job/devel_job_db_noetic_ros_controllers.sh
cd /tmp/devel_job
sh devel_job_db_noetic_ros_controllers.sh
```

sloretz, 2020-12-08:
@rhaschke I opened a pull request to solve the fcl issue: https://github.com/flexible-collision-library/fcl/pull/514 . Mind making another release when it lands?

This time I verified it fixes the issue using the `generate_release_script.py` docker images. I saw CMake does think the host supports SSE2, and checking compiler support directly does fix the build.

tylerjw, 2020-12-08:
Thank you for fixing fcl.  :grinning: 

We are currently dealing with some unrelated unstable testing issues in MoveIt.  Once we resolve those issues I'm happy to make a Noetic release of MoveIt also after FCL is released.

rhaschke, 2020-12-08:
Sure, I will prepare a new release as soon as the PR is merged. Thanks for working on this!

hidmic, 2021-02-12:
@sloretz can we call this done?

sloretz, 2021-03-04:
It almost can be closed. `ros_controllers` has since been re-released and appears to be building. `moveit` is still rolled back to 1.1.0. I opened a PR to re-release it in #28486

DLu, 2021-06-11:
Now is it done?

torgeirl, 2021-06-30:
> As far as I can tell there are still issues the appeared with this regression, that are unsolved. For instance, Gazebo is no longer are able to run `turntable.urdf.xacro` in ROS simulations.

Not sure what got addressed the last few weeks, but at least the issue I referenced has recently been resolved. :tada:

sloretz, 2021-08-12:
I think this can be closed :tada:

closed 2021-08-12 by sloretz