ros/rosdistro #27538: ROS kinetic pcl dependency conflict

ivan1993br, 2020-12-01:
Hi guys,

There was a new release of `libvtk6-dev` on [ubuntu xenial](http://changelogs.ubuntu.com/changelogs/pool/universe/v/vtk6/vtk6_6.2.0+dfsg1-10ubuntu0.1/changelog), that conflicts with the osrf one, and in consequence with `ros-kinetic-pcl-ros` package. 

The osrf release for vtk6 is `6.2.0+dfsg1-10build1+debian11.1+osrf1`.

As the "most updated" version now, is the `ubuntu updates` one we are not installing the osrf version.

Can we have an update on this lib?

Thanks!

tfoote, 2020-12-02:
For history on the imported version see: https://github.com/ros-perception/perception_pcl/pull/124  and https://github.com/ros-infrastructure/reprepro-updater/issues/32

The changelog looks like exactly what we patched in and was contributed by @kyrofa and was the same minimal change thus I would have hoped that the new upstream package would be a good replacement.

kyrofa, 2020-12-02:
As far as I understand it, it may not quite be the same. I believe you followed the Debian reversion back to the vtk-vendored libproj, right? We can't do that in Ubuntu as it would have broken ABI, so I tried to fix it by instead pushing all the way through to using the system libproj by adding the dep and rebuilding libpcl to have the proper library in its cmake config files.

ivan1993br, 2020-12-02:
So, if I understood correctly, we used a version of the pcl release that used the vtk-vendored libproj, the recent update on PCL fix the ABI and uses the system libproj, I was able to "fix" just setting the vtk6, using the system libpcl, but the vtk-vendored libproj is dangling there and not being used by pcl, just linked and not used on catkin packages.

We have two options for now: new release on vtk6 and pcl on top of the new one, or rebuild all ros packages dependent on PCL and their dependencies.

What about just releasing a new vtk6, but leave pcl as it is, leaving vtk-vendored libproj dangling? Too fragile? This would allow to update the packages libraries as we go. Does it make sense?

kyrofa, 2020-12-02:
Yeah, to clarify the state of Ubuntu before the bugfix was:

- libvtk6 has a vendored copy of libproj that it used to use. When it was being used, it was called, sensibly, vtkproj4. However, the version in Xenial doesn't use that vendored copy, and thus doesn't build it, linking instead to the system libproj. As a result, it didn't provide a library named vtkproj4
- libvtk6, despite building against and requiring libproj, did not add it to its dependency list
- libpcl-dev was built back when libvtk6 WAS using its vendored libproj, and thus had vtkproj4 written into its PCLConfig.cmake files

As a result, anyone using libvtk6 ended up with errors because they were using a library that required libproj to be installed, but it didn't depend on it. Also, anyone using cmake to link to pcl ended up with build errors because they were trying to link to a library that didn't exist. It was just all-around broken.

The fix that Open Robotics got into their Debian archive was to modify libvtk6 to start building and using its vendored libproj again, which made the libpcl in Xenial work because it could find that vtkproj4 library again. However, the vendored libproj is not ABI compatible with the system libproj, so that's not something I could safely make happen in Xenial. I instead made libvtk6 properly depend upon libproj, and then rebuilt libpcl so it didn't have a library that didn't exist in its cmake config.

> We have two options for now: new release on vtk6 and pcl on top of the new one, or rebuild all ros packages dependent on PCL and their dependencies.

That's my thinking anyway, yes.

> What about just releasing a new vtk6, but leave pcl as it is, leaving vtk-vendored libproj dangling? Too fragile? This would allow to update the packages libraries as we go. Does it make sense?

No, I think that will break. It's exactly the opposite situation of the breakage that was in Xenial before this happened:

- libvtk will be using its vendored libproj
- libpcl will be using the system libproj

I suspect that will be one big segfault for anyone trying to use libpcl, because as I mentioned, the vendored libproj and the system libproj are not ABI-compatible.

tfoote, 2020-12-02:
I just talked with @kyrofa and since the upstream libvtk and libpcl are both now consistent the simplest solution is to do the general rebuild of Kinetic to use the consistent libproj. Although it's not an ABI break, the solution is similar, we will rebuild all packages in Kinetic depending on libvtk6 and anyone with compiled binary distributions will have to do the same. This will take some server time but will provide the simplest solution in the long run.

Process:

1. [ ] rebuild all kinetic packages on top of the new libvtk6
2. [ ] validate the fix in shadow-fixed
3. [ ] Sync and notify all users to do a dist-upgrade to make sure to get all the new binaries, and recommend all binary distributed content to be recompiled as well.

nuclearsandwich, 2020-12-03:
I'm in favor of the above plan. Do we want to scrub the vendored vtk from our repositories once it is no longer needed or leave it up for the forensics of others?

ivan1993br, 2020-12-03:
Thank @tfoote!

For those in need of a temporary fix now [ppa](https://launchpad.net/~ivan-tarifa/+archive/ubuntu/vtk6-osrf-temporary-release). (It may be removed after the binary distribution)

Currently only with vtk6, I executed integration tests on the robot (it has some custom layers, sonar pcl), and did not break, but as pointed @kyrofa  the ABI is not compatible. I do not know if we are missing something.

The ideas was just to do some tests with this partial fix, and then push the PCL, but launchpad today was rather slow for some reason.

tfoote, 2020-12-03:
As a temporary fix for now I would recommend just pinning back the version of vtk as outlined on this issue: https://github.com/ros-perception/perception_pcl/issues/318

> Do we want to scrub the vendored vtk from our repositories once it is no longer needed or leave it up for the forensics of others?

After this patch is out it would no longer be necessary for anything rebuilt, but I'd suggest leaving it in case someone needs to pin it for a distributed binary from before the patch.

kyrofa, 2020-12-03:
> I'd suggest leaving it in case someone needs to pin it for a distributed binary from before the patch.

Seconded. No harm in leaving other than space, but definitely a potential benefit, there.

tfoote, 2020-12-03:
I'm seeing a problem of the rebuild on armhf: http://build.ros.org/view/Ksrc_uX/job/Kbin_uxhf_uXhf__pcl_ros__ubuntu_xenial_armhf__binary/16/console

Console error Error: `18:08:14 <command-line>:0:15: warning: missing whitespace after the macro name`
<details>

```
18:06:13 make[4]: Entering directory '/tmp/binarydeb/ros-kinetic-pcl-ros-1.4.4/obj-arm-linux-gnueabihf'
18:06:13 [  1%] Building CXX object CMakeFiles/pcl_ros_tf.dir/src/transforms.cpp.o
18:06:13 /usr/lib/ccache/arm-linux-gnueabihf-g++   -DDISABLE_OPENNI2 -DDISABLE_PCAP -DDISABLE_PNG -DROSCONSOLE_BACKEND_LOG4CXX -DROS_BUILD_SHARED_LIBS=1 -DROS_PACKAGE_NAME=\"pcl_ros\" -Dpcl_ros_tf_EXPORTS -DvtkFiltersFlowPaths_AUTOINIT="1(vtkFiltersParallelFlowPaths)" -DvtkIOExodus_AUTOINIT="1(vtkIOParallelExodus)" -DvtkIOGeometry_AUTOINIT="1(vtkIOMPIParallel)" -DvtkIOImage_AUTOINIT="1(vtkIOMPIImage)" -DvtkIOSQL_AUTOINIT="2(vtkIOMySQL,vtkIOPostgreSQL)" -DvtkRenderingContext2D_AUTOINIT="1(vtkRenderingContextOpenGL)" -DvtkRenderingCore_AUTOINIT="4(vtkInteractionStyle,vtkRenderingFreeType,vtkRenderingFreeTypeOpenGL,vtkRenderingOpenGL)" -DvtkRenderingFreeType_AUTOINIT="2(vtkRenderingFreeTypeFontConfig,vtkRenderingMatplotlib)" -DvtkRenderingLIC_AUTOINIT="1(vtkRenderingParallelLIC)" -DvtkRenderingVolume_AUTOINIT="1(vtkRenderingVolumeOpenGL)" -I/tmp/binarydeb/ros-kinetic-pcl-ros-1.4.4/obj-arm-linux-gnueabihf/devel/include -I/usr/include/vtk-6.2 -I/usr/include/jsoncpp -I/usr/include/arm-linux-gnueabihf -I/usr/include/freetype2 -I/usr/include/arm-linux-gnueabihf/freetype2 -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent -I/usr/lib/openmpi/include/openmpi/opal/mca/event/libevent2021/libevent/include -I/usr/lib/openmpi/include -I/usr/lib/openmpi/include/openmpi -I/usr/include/libxml2 -I/usr/include/hdf5/serial -I/usr/include/python2.7 -I/usr/include/tcl -I/tmp/binarydeb/ros-kinetic-pcl-ros-1.4.4/include -I/opt/ros/kinetic/include -I/opt/ros/kinetic/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -I/usr/include/eigen3 -I/usr/include/pcl-1.7 -I/usr/include/ni  -g -O2 -fstack-protector-strong -Wformat -Werror=format-security -DNDEBUG -Wdate-time -D_FORTIFY_SOURCE=2   -fPIC   -DDISABLE_LIBUSB-1.0 -o CMakeFiles/pcl_ros_tf.dir/src/transforms.cpp.o -c /tmp/binarydeb/ros-kinetic-pcl-ros-1.4.4/src/transforms.cpp
18:08:14 <command-line>:0:15: warning: missing whitespace after the macro name
18:08:14 make[4]: *** No rule to make target '/usr/lib/arm-linux-gnueabihf/hdf5/openmpi/lib/libhdf5.so', needed by 'devel/lib/libpcl_ros_tf.so'.  Stop.
18:08:14 make[4]: Leaving directory '/tmp/binarydeb/ros-kinetic-pcl-ros-1.4.4/obj-arm-linux-gnueabihf'
18:08:14 CMakeFiles/Makefile2:3698: recipe for target 'CMakeFiles/pcl_ros_tf.dir/all' failed
18:08:14 make[3]: *** [CMakeFiles/pcl_ros_tf.dir/all] Error 2
```
</details>

kyrofa, 2020-12-03:
That's from perception_pcl? It builds locally, it sounds like a dep hasn't been rebuilt yet. I think it's related to the dependency your patched version had that the one in Xenial does not have:

```
$ apt-cache policy libvtk6.2
libvtk6.2:
  Installed: 6.2.0+dfsg1-10ubuntu0.1
  Candidate: 6.2.0+dfsg1-10ubuntu0.1
  Version table:
 *** 6.2.0+dfsg1-10ubuntu0.1 500    # <- Xenial's updated version
        500 http://archive.ubuntu.com/ubuntu xenial-updates/universe amd64 Packages
        100 /var/lib/dpkg/status
     6.2.0+dfsg1-10build1+debian11.1+osrf1 500  # <- OR's version
        500 http://packages.ros.org/ros/ubuntu xenial/main amd64 Packages
     6.2.0+dfsg1-10build1 500  # <- Old, broken version
        500 http://archive.ubuntu.com/ubuntu xenial/universe amd64 Packages

# Old version:
$ apt-cache depends libvtk6.2=6.2.0+dfsg1-10build1 | grep mpi
  Depends: libopenmpi1.10
  Suggests: mpi-default-bin
$ apt-cache depends libvtk6-dev=6.2.0+dfsg1-10build1 | grep mpi
  Depends: mpi-default-dev

# Xenial's (updated) version:
$ apt-cache depends libvtk6.2 | grep mpi
  Depends: libopenmpi1.10
  Suggests: mpi-default-bin
$ apt-cache depends libvtk6-dev | grep mpi
  Depends: mpi-default-dev

# OR's version:
$ apt-cache depends libvtk6.2=6.2.0+dfsg1-10build1+debian11.1+osrf1 | grep mpi
  Depends: libhdf5-openmpi-10
  Depends: libopenmpi1.10
  Suggests: mpi-default-bin
$ apt-cache depends libvtk6-dev=6.2.0+dfsg1-10build1+debian11.1+osrf1 | grep mpi
  Depends: libhdf5-mpi-dev
  Depends: mpi-default-dev
```

If you perception_pcl depends upon it, it should be explicit, but I suspect it's coming from another cmake config file somewhere.

tfoote, 2020-12-04:
Ahh after some investigation we are hosting a backport of libpcl for armhf only that has the dependency on vtk6.2 and thus needs to be rebuilt to clear the cached cmake config files passing through the vtk dependencies. http://repositories.ros.org/ubuntu/building/dists/xenial/main/binary-armhf/Packages

The backports are [pulled here](https://github.com/ros-infrastructure/reprepro-updater/blob/8225aa1095ce2ce67218d469bd96f6b2dc84bef6/config/pcl_xenial_armhf_backport.yaml#L2) 
 https://launchpad.net/~tully.foote/+archive/ubuntu/backports/+packages

Which was a workaround for #11583 

It looks like the backports will need to be rebuilt and reimported.

kyrofa, 2020-12-04:
@tfoote that's https://bugs.launchpad.net/ubuntu/+source/pcl/+bug/1906277 . It's fixed in -proposed and is in the 7-day waiting period, should be released to -updates on the 10th.

tfoote, 2020-12-04:
Thanks for the heads up. I'd much rather use the upstream one. With everything broken right now I don't want to wait a full week, or the remainder of the week. We should be able to import the proposed packages by updating this url:

https://github.com/ros-infrastructure/reprepro-updater/blob/8225aa1095ce2ce67218d469bd96f6b2dc84bef6/config/pcl_xenial_armhf_backport.yaml#L2

So we don't diverge again from upstream and require ongoing maintenance.

kyrofa, 2020-12-04:
Yeah sounds perfect.

tfoote, 2020-12-05:
With the xenial-proposed packages for armhf we're now approaching a complete rebuild. I expect it will be complete in the morning.

tfoote, 2020-12-07:
The sync is out: https://discourse.ros.org/t/new-packages-for-kinetic-kame-2020-12-07/17786

closed 2020-12-07 by tfoote