ros/rosdistro #27851: Inconsistent Gazebo versions in Debian Buster

StefanFabian, 2020-12-28:
I've encountered this issue in this [failed build log](https://build.ros.org/job/Nbin_db_dB64__gazebo_ros_control_select_joints__debian_buster_amd64__binary/6/console) on Debian Buster.
Apparently, the gazebo rosdep key on Debian Buster resolves to gazebo9
```bash
$ rosdep resolve --os=debian:buster gazebo
#apt
gazebo9
```
However, the noetic version of gazebo-dev `ros-noetic-gazebo-dev` for [Debian Buster](http://packages.ros.org/ros/ubuntu/dists/buster/main/binary-amd64/Packages) depends on `gazebo11`.

What is the recommended approach for dealing with this?
While in theory it would possibly break some Debian Buster setups to switch the `gazebo` key to `gazebo11`, I'd argue that it already is broken since some packages depend on `gazebo11` and others via `gazebo` on `gazebo9`.
There are no debian packages available for Buster that depend on gazebo9 as far as a simple search could tell (5 matches all Conflicts declarations).

StefanFabian, 2021-01-06:
@tfoote @wjwwood @sloretz
Any opinion on this?
Should I just submit a patch for buster that changes the key to gazebo11?
Seems like a hacky solution which is why I haven't done it so far.

tfoote, 2021-02-02:
Yeah, this is kind of ending up in a noman's land where there are two versions of gazebo available on that platform that conflict. I think that since this is the keys for the ROS installation it makes sense to target the version that ROS is expecting. We ran into this problem even more so with stretch where the stretch was officially targeted by two distros with conflicting version requirements. (#19843) 

I think that homogenizing these rules to version 11 makes sense on buster to make sure that we're consistent and don't end up with people having incompatible dependencies.

closed 2021-05-20 by mabelzhang

StefanFabian, 2021-05-27:
Hm, the build is still failing. From the docker commands, I read that the install command is cached?
Do I have to make another release to make sure it uses the new value?

commit f859d33e (2021-05-20): Update default gazebo version on debian buster (#29578)

Fixes #27851