ros/rosdistro #28454: Missing rosdep keys for eloquent

rhaschke, 2021-03-02:
Trying to release xacro for Eloquent, I'm getting these errors:
```
Pre-verifying Debian dependency keys...
Running 'rosdep update'...
Could not resolve rosdep key 'ament_cmake'
Failed to resolve ament_cmake on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'ament_cmake', aborting.
ament_cmake is depended on by these packages: ['xacro']
<== Failed
Could not resolve rosdep key 'ament_cmake_pytest'
Failed to resolve ament_cmake_pytest on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'ament_cmake_pytest', aborting.
ament_cmake_pytest is depended on by these packages: ['xacro']
<== Failed
Could not resolve rosdep key 'ament_index_python'
Failed to resolve ament_index_python on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'ament_index_python', aborting.
ament_index_python is depended on by these packages: ['xacro']
<== Failed
Could not resolve rosdep key 'ament_lint_auto'
Failed to resolve ament_lint_auto on ubuntu:bionic with: Error running generator: Failed to resolve rosdep key 'ament_lint_auto', aborting.
ament_lint_auto is depended on by these packages: ['xacro']
<== Failed
Some of the dependencies for packages in this repository could not be resolved by rosdep.
<== The following generator action reported that it is missing one or more
    rosdep keys, but that the key exists in other platforms:
'['/usr/bin/git-bloom-generate', '-y', 'rosdebian', '--prefix', 'release/eloquent', 'eloquent', '-i', '1', '--os-name', 'ubuntu']'

If you are absolutely sure that this key is unavailable for the platform in
question, the generator can be skipped and you can proceed with the release.
```

clalancette, 2021-03-02:
Eloquent is now EOL, so we shouldn't release updates for it anyway.

closed 2021-03-02 by clalancette

rhaschke, 2021-03-02:
Thanks for this reminder, @clalancette. I just learned about the rolling release as well (I don't yet actively use ROS2).
As far as I understand, I should release `xacro` into `rolling` too, shouldn't I? 
Trying to do so, I'm missing permissions to the ROS2 release repo, which is different from the one used so far.
```
remote: Permission to ros2-gbp/xacro-release.git denied to rhaschke.
fatal: unable to access 'https://github.com/ros2-gbp/xacro-release.git/': The requested URL returned error: 403
Cannot push to remote release repository.
```
Who is going to grant that permission if I should get it?

clalancette, 2021-03-02:
> As far as I understand, I should release `xacro` into `rolling` too, shouldn't I?

Yeah, releasing it into Rolling would be best.  That way you'll find out much earlier if there are any upcoming changes that you may need to make.

> Who is going to grant that permission if I should get it?

Well, I can certainly give you the permissions.  However, I would like to get confirmation from @nuclearsandwich that we should be using https://github.com/ros2-gbp/xacro-release for this.