ros/rosdistro #28891: Pip entries

MatthijsBurgh, 2021-03-29:
The current guidelines say that `python-*-pip` entries don't get a `python3-*-pip` version of that entry. (Confirmed by @tfoote in https://github.com/ros/rosdistro/pull/28264#issuecomment-784463002)

But also that new pip entries should use the `python3` prefix (https://github.com/ros/rosdistro/pull/28672#pullrequestreview-614757890) This will result in a big mess of `python-*-pip` and `python3-*-pip` entries. This is not straightforward.

I suggest we either allow the creation of `python3` version of current `python-*-pip` entries, so that in the end all entries exist as a `python3` version and the `python` ones get outdated.

Of we should stick to `python-*-pip` as pip targets are the same for python2 and python3.

Let's clear this up and prevent a big mess.

tfoote, 2021-04-07:
Unfortunately this is a case where there's no clear cut winner. 

It would generally be very nice to have all rules following the new standard. However if there are people already using the rules we generally do not want to break them. So our default is to keep existing rules unless they're wrong or broken. If we can we also make sure to phase things out so that we don't take away things that are currently working, but change rules to not work on upcoming platforms as a somewhat graceful way to migrate without breaking a running system.

We will then have to go through all the keys and implement the sunsetting logic on the keys, disabling it on future platforms. And then every package maintainer will be required to change their codebase to switch over as well. The scope of this mandatory required change is hard to measure, but quite large.

To avoid the costs of switching we could attempt to maintain both in parallel, but then that will just lead to confusion as well. And that makes the maintenance much higher which is a cost we really don't want to incur.

We've already had several cases where using the python3 prefix has successfully prevented collisions with apt based rules. The value in this is mostly on insertion of new rules. Which doesn't have the same benefits for existing packages. And the native packages are a much higher priority as they can be used to package against than the pip packages which are much less QAd.

As much has having a few oddball legacy keys with the old prefix isn't the most obvious. When a user is looking for a key to use, that user usually search for the python library name and this will find it regardless of the prefix. In a file with 10k items I find it unlikely that the user will not use at least a text search in their browser. If not using the online index search: https://index.ros.org/search/?term=jetson&section=deps which makes it unlikely that people will be unable to find the other rule.

However it might be possible to do aliases which will avoid the double maintenance. Also we are adding more auto validation tools which might be able to detect the duplicate rules. But otherwise we'll likely stick with the existing legacy keys and not actively migrate them. Over time the fraction of the database with these legacy behaviors will drop away. It's not perfect but the cost of finding the other rule is pretty low compared to breaking an unknowably large number of code bases.

tfoote, 2021-04-07:
Actually with a little bit of experimentation I think we can leverage the alias ability in yaml to keep legacy rules w/o doubling the overhead burden.

```
python3-autograd-pip: &legacy_autograd_pip
  debian:
    pip:
      packages: [autograd]
  fedora:
    pip:
      packages: [autograd]
  ubuntu:
    pip:
      packages: [autograd-new-rule]

autograd-pip: *legacy_autograd_pip
```

Adding the above manually to my rosdep sources. 

```
tfoote@snowman5:~ Last: [0] (7s Seconds)
$ rosdep resolve python3-autograd-pip
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
#pip
autograd-new-rule
tfoote@snowman5:~ Last: [0] (0s Seconds)
$ rosdep resolve autograd-pip
WARNING: ROS_PYTHON_VERSION is unset. Defaulting to 3
#pip
autograd-new-rule
```

If we do this I'd suggest that we build in an EOL date for the redirects and embed that EOL date into the anchor so that we can know when its ok to clean it out. If we pick an EOL date of ~5 years or the end of the longest supported current platform then we can make sure not to delete. Someting of the format `legacy_eol_2025_04_30_<OLD_KEY_UNDERSCORED>`

MatthijsBurgh, 2021-04-07:
@tfoote, this sounds like a very good solution.

MatthijsBurgh, 2021-04-08:
@tfoote I think that the anchor needs to be defined before the alias is used. To keep the alphabetical order of the entries (So `python3-` comes after `python-`), we need to switch the anchor and alias around. So the `python-*-pip` entry contains the content and the anchor. While the `python3-*-pip` entry becomes the alias.

```yaml
python-foobar-pip: &legacy_eol_2025_04_30_python_foobar_pip
  ubuntu:
    pip:
      packages: [foobar]
python3-foobar-pip: *legacy_eol_2025_04_30_python_foobar_pip
```

This is a little bit of a downside as dropping the legacy entries in the future means moving them.

In this case a different format of the anchor/alias might be better. Something which inlcudes `migrating` or similar IMO.

MatthijsBurgh, 2021-04-08:
@tfoote I have made a proposal in #29037

PR #29037 (2021-05-06): (contributing) Add guidelines for anchoring and aliasing of pip entries
As discussed in #28891

Fixes #28891
closed 2021-05-06 by nuclearsandwich