ros/rosdistro #28910: How to add key for Jetson stats package

Prat33k-dev, 2021-03-31:
Earlier my teammate @maarufvazifdar  added jetson-stats package [](https://pypi.org/project/jetson-stats/) but found that it is not working with rosdep. Also to install the package manually "pip install jetson-stats" does not work but "sudo -H pip install -U jetson-stats" works. So, how should I define it in the python.yaml file?
Reference for old PR- [](https://github.com/ros/rosdistro/pull/28280)

tfoote, 2021-04-07:
Without more information "not working with rosdep" is hard for us to say what the problem is. If rosdep is invoking pip install and the installation is failing then you'll need to look into the package itself. It may not be mature enough to work consistently.

We don't support using the `-H` directive. That's relating to using the user's home directory which makes it user account specific which is not expected for a system installation. I tested it without the -H option and it seemed to install just fine. You also say that pip install doesn't work without the sudo. These both seem like issues to take up with the upstream package and are not something that we can help with.

Prat33k-dev, 2021-04-09:
Ok, we will look into it. Thanks for the updates

closed 2021-04-09 by Prat33k-dev