ros/rosdistro #29168: "Failed to open pull request" with HTTP Error 404

ROBOTIS-Will, 2021-04-15:
Hi, 

After enabling GitHub 2FA for my account, and applying [manually authorizing bloom to generate PR](http://wiki.ros.org/bloom/Tutorials/GithubManualAuthorization), my PR to rosdistro starts failing with 404 error.
```
Failed to open pull request: GithubException - HTTP Error 404: Not Found (https://api.github.com/repos/ros/rosdistro/pulls): None
```
Is this something have to do with the 2FA?

ROBOTIS-Will, 2021-04-16:
I found an identical issue on ros-infrastructure/bloom
https://github.com/ros-infrastructure/bloom/issues/632
Closing this. Thanks!

closed 2021-04-16 by ROBOTIS-Will

Kukanani, 2021-04-16:
Just wanted to follow up on this and note that I don't have 2FA on my account, so I doubt that is the source of the issue.