ros/rosdistro #29173: File conflict for /opt/ros/noetic/lib/python3/dist-packages/libgtest.so between moveit_core and tf

rhaschke, 2021-04-15:
Since the MoveIt release was merged recently in #29063, we have a [file conflict reported on the buildfarm](https://build.ros.org/view/Nbin_uF64/job/Nbin_uF64__moveit_ros_planning__ubuntu_focal_amd64__binary/25/console):
```
08:11:30   2967 'dpkg: error processing archive /tmp/apt-dpkg-install-UN3zrI/632-ros-noetic-tf_1.13.2-1focal.20210414.184309_amd64.deb (--unpack):'
08:11:30   2968 ' trying to overwrite '/opt/ros/noetic/lib/python3/dist-packages/libgtest.so', which is also in package ros-noetic-moveit-core 1.1.2-1focal.20210414.202258'
```
IMO, neither of those packages should install `libgtest.so`! Any ideas how this enters the debian package?

rhaschke, 2021-04-15:
@tylerjw for your info

tylerjw, 2021-04-15:
This is really odd.  I don't understand how the install into `/opt/ros/noetic/lib/python3/dist-packages/` works and how we impact that.  Could this be something coming from a rosdep install run?  I don't think anything we are doing should be manually installing anything into there.

I'm not sure how to debug this.

rhaschke, 2021-04-15:
Of course, we don't install anything manually in /opt/ros. This is done by the debian package that is built from our ROS packages. However, I haven't seen any install directive for `libgtest` in our sources. @sloretz,  has anything changed to catkin or the debian release process that could explain this issue?

k-okada, 2021-04-16:
@tylerjw, @rhaschke I have same problem, with `pr2-controller-manager` and `moveit-core`
```
Preparing to unpack .../ros-noetic-pr2-controller-manager_1.8.18-1focal.20210414.185725_amd64.deb ...
Unpacking ros-noetic-pr2-controller-manager (1.8.18-1focal.20210414.185725) ...
dpkg: error processing archive /var/cache/apt/archives/ros-noetic-pr2-controller-manager_1.8.18-1focal.20210414.185725_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/noetic/lib/python3/dist-packages/libgtest.so', which is also in package ros-noetic-moveit-core 1.1.2-1focal.20210414.202258
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
Errors were encountered while processing:

```

rhaschke, 2021-04-16:
It looks like `setup.py` is installing `libgtest.so`:
```
+ mkdir -p /homes/rhaschke/src/ros/install/lib/python3/dist-packages
+ /usr/bin/env PYTHONPATH=/homes/rhaschke/src/ros/install/lib/python3/dist-packages:/homes/rhaschke/src/ros/build/tf/lib/python3/dist-packages:/opt/ros/noetic/lib/python3/dist-packages CATKIN_BINARY_DIR=/homes/rhaschke/src/ros/build/tf /usr/bin/python3 /homes/rhaschke/src/ros/src/ros/tf/tf/setup.py egg_info --egg-base /homes/rhaschke/src/ros/build/tf build --build-base /homes/rhaschke/src/ros/build/tf install --root=/ --install-layout=deb --prefix=/homes/rhaschke/src/ros/install --install-scripts=/homes/rhaschke/src/ros/install/bin
running egg_info
... 
running build
running build_py
...
running build_scripts
...
running install
running install_lib
creating /homes/rhaschke/src/ros/install/lib/python3/dist-packages/tf
copying /homes/rhaschke/src/ros/build/tf/lib/tf/broadcaster.py -> /homes/rhaschke/src/ros/install/lib/python3/dist-packages/tf
copying /homes/rhaschke/src/ros/build/tf/lib/tf/__init__.py -> /homes/rhaschke/src/ros/install/lib/python3/dist-packages/tf
copying /homes/rhaschke/src/ros/build/tf/lib/tf/listener.py -> /homes/rhaschke/src/ros/install/lib/python3/dist-packages/tf
copying /homes/rhaschke/src/ros/build/tf/lib/tf/transformations.py -> /homes/rhaschke/src/ros/install/lib/python3/dist-packages/tf
copying /homes/rhaschke/src/ros/build/tf/lib/tf/tfwtf.py -> /homes/rhaschke/src/ros/install/lib/python3/dist-packages/tf
copying /homes/rhaschke/src/ros/build/tf/lib/libgtest.so -> /homes/rhaschke/src/ros/install/lib/python3/dist-packages
```
@sloretz, is it possible that `setup.py` just installs all libs from `<build space>/<package>/lib` (libgtest.so is the only lib there) and this has changed?

v4hn, 2021-04-16:
We already debugged this some time ago and it's a problem in gtest's cmake files that can be worked around in catkin: https://github.com/ros/catkin/pull/1135

rhaschke, 2021-04-16:
@mjcarroll, could you please have a look at https://github.com/ros/catkin/pull/1135 to unblock the upcoming Noetic release?

peci1, 2021-04-20:
The catkin issue has been merged, so this one can probably be closed.

rhaschke, 2021-04-20:
@mjcarroll, I guess catkin requires a new release before we can proceed with other packages?

jacobperron, 2021-06-15:
What's the status of this ticket? It looks there's been a release of catkin since https://github.com/ros/catkin/pull/1135.

k-okada, 2021-06-16:
@jacobperron my problem on pr2 packages has been solved, thank you!
https://github.com/PR2/pr2_controllers/pull/401, https://github.com/PR2/pr2_kinematics/pull/15

rhaschke, 2021-06-16:
Yes, problem is solved.

closed 2021-06-16 by rhaschke