ros/rosdistro #29398: version number confusion

swhart115, 2021-05-03:
So I'm confused.  I have a package (trac_ik), that i'm trying to get fixed for noetic. I didnt originally set this all up for bloom, so i'm still trying to make sense of everything. 

I noticed that if i run the release scripts to release a new version from
http://wiki.ros.org/bloom/Tutorials/ReleaseCatkinPackage
that the prepare script bumps the minor number in the package.xmls (currently they are at 1.6.4, running it again would put the packages at 1.6.5)

But if i run the bloom-release script, it is only bumping the patch number.  this means they are out of sync, and the config file is calling it 1.6.1-8.. If i ran it again it would make it 1.6.1-9.  

so i feel like i'm missing something cause this is inconsistent.  what am i misunderstanding?

clalancette, 2021-05-03:
> so i feel like i'm missing something cause this is inconsistent. what am i misunderstanding?

Usually in these cases it is that the `devel_branch` in the release repository is pointing to the wrong place.  Ah, but looking at https://github.com/traclabs/trac_ik-release/blob/master/tracks.yaml , it is actually a different problem.  Because the version is "pinned" at 1.6.1, it won't ever do another release.

My suggestion here is to use `bloom-release --edit --track noetic --rosdistro noetic trac_ik`, and switch the "version" to `:{auto}` (like is done for Melodic).  That will make bloom automatically look at the source repository and try to figure out the release number from the currently configured `devel_branch`.

swhart115, 2021-05-03:
ok, great. that makes more sense.  let me run the script that way.  i just made sure that the package.xmls have the same version as the git tag on the repo...

swhart115, 2021-05-03:
yeah, things seem to be in sync now: https://github.com/ros/rosdistro/pull/29399

Thanks for the help!

closed 2021-05-03 by swhart115