ros/rosdistro #29465: ROS_DISTRO is not defined in foxy

doronhi, 2021-05-09:
Merged #29445 I see that the buildfarm keeps failing for the following error:

09:15:35 CMake Error at CMakeLists.txt:119 (message):
09:15:35   ROS_DISTRO is not defined.

Caused by the following lines in the CMakelists.txt file:
```
if(NOT DEFINED ENV{ROS_DISTRO})
  message(FATAL_ERROR "ROS_DISTRO is not defined." )
endif()
```

(See full log [here:](https://build.ros2.org/job/Fbin_ubv8_uFv8__realsense2_camera__ubuntu_focal_arm64__binary/132/consoleFull) )

Is it a syntax error on my behalf?
Isn't ROS_DISTRO supposed to be defined?
I saw [this](https://answers.ros.org/question/277124) but it seems to apply to ROS as roslib is not a ROS package.

Also, is there a docker image I can use to test my package.xml and CMakelists.txt files?

clalancette, 2021-05-09:
ROS_DISTRO is defined by the `ros-environment` package, so you'll only have it set on the buildfarm if you explicitly have a `build_depend` on `ros_environment`.  I think that should fix your immediate problem.

doronhi, 2021-05-10:
Thanks! Fixed and submitted as #29473

closed 2021-05-10 by nuclearsandwich