ros/rosdistro #29715: Expired certificate (2021-05-29)

eigendude, 2021-05-29:
## Description

Today, updating via `sudo apt update` started giving the error:

```
Err:3 http://packages.ros.org/ros2/ubuntu bionic InRelease
  The following signatures were invalid: EXPKEYSIG F42ED6FBAB17C654 Open Robotics <info@osrfoundation.org>
```

It looks like the certificate just expired:

```
$ sudo apt-key list | grep -A 1 expired

Warning: apt-key output should not be parsed (stdout is not a terminal)
pub   rsa4096 2019-05-30 [SC] [expired: 2021-05-29]
      C1CF 6E31 E6BA DE88 68B1  72B4 F42E D6FB AB17 C654
uid           [ expired] Open Robotics <info@osrfoundation.org>
```

Is it possible to generate a new cert?

kscottz, 2021-05-29:
We're aware of the issue and we're presently taking steps to resolve the issue. There is an [incident report available](https://status.ros.org/incidents/mhqz8yyhrtsg) and a [discourse thread](https://discourse.ros.org/t/key-for-downloading-ros2-source-showing-expired/20662/2) where we'll post updates.

dongpan90, 2021-05-29:
@eigendude  
I met the same problem. How did you solve it?

kscottz, 2021-05-29:
We're working on a solution presently. We should have a remedy within the next few hours.

zmk5, 2021-05-29:
> We're working on a solution presently. We should have a remedy within the next few hours.

Thanks Kat!

rocketsfallonrocketfalls, 2021-05-29:
For people who needs an urgent/quick fix, you can try forcing insecure apt update according to a post from today on: https://answers.ros.org/question/379190/apt-update-signatures-were-invalid-f42ed6fbab17c654/

kscottz, 2021-05-29:
[We have posted an incident report and the steps needed to fix the issue on ROS Discourse. ](https://discourse.ros.org/t/ros-gpg-key-expiration-incident/20669)

realtime-neil, 2021-05-29:
Wait. Do I understand correctly that the new key has the exact same ID as the old key? That they differ only by expiration date?

realtime-neil, 2021-05-29:
@tfoote In the (very likely) case that it is impractical to "update" the keyservers with new content under the existing ID, would you kindly consider creating a new key? If there's a keyserver that supports "overwriting" a given ID, then I haven't (yet) found one.

realtime-neil, 2021-05-29:
Nevermind; I got it:

```
$ gpg --keyserver keyserver.ubuntu.com --send-keys C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
gpg: sending key F42ED6FBAB17C654 to hkp://keyserver.ubuntu.com
$ echo $?
0
```

```
$ date -uIseconds && curl -fsSL "http://keyserver.ubuntu.com/pks/lookup?op=get&search=0xc1cf6e31e6bade8868b172b4f42ed6fbab17c654" | gpg --import --import-options show-only
2021-05-29T15:48:41+00:00
pub   rsa4096 2019-05-30 [SC] [expires: 2025-06-01]
      C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654
uid                      Open Robotics <info@osrfoundation.org>

gpg: Total number processed: 1
```

TIL that you can update a key on a keyserver.

eigendude, 2021-05-30:
All good on my end. Thanks for the quick response!

closed 2021-05-30 by eigendude