ros/rosdistro #29738: bloom release galactic

JanStaschulat, 2021-06-01:
Hi, I have a question regarding executing the bloom release for Galactic. In a pull request to rclc  https://github.com/ros2/rclc/pull/94, the ci-job for the Galactic bloom release fails, because the branch `origin/galactic^{commit}` was not found:
```
23:36:38  > git rev-parse origin/galactic^{commit} # timeout=10
23:36:38 FATAL: Command "git rev-parse origin/galactic^{commit}" returned status code 128:
23:36:38 stdout: origin/galactic^{commit}
23:36:38 
23:36:38 stderr: fatal: ambiguous argument 'origin/galactic^{commit}': unknown revision or path not in the working tree.
```

https://build.ros2.org/job/Gpr__rclc__ubuntu_focal_amd64/47/

We are using `ros-tooling/setup-ros@0.2.1` and `ros-tooling/action-ros-ci@0.2.1`.

I updated the branch version from `master` to `galactic` in this [PR](https://github.com/ros/rosdistro/pull/29683)
Then, I updated the version also with the bloom-release tool : https://github.com/ros/rosdistro/pull/29702
While creating the bloom release using the command: 

`bloom-release --rosdistro galactic --edit-track --track galactic rclc`, 

I  corrected the `Upstream Devel Branch` to `galactic`: 
```
Version:
  :{ask}
    This means that the user will be prompted for the version each release.
    This also means that the upstream devel will be ignored.
  :{auto}
    This means the version will be guessed from the devel branch.
    This means that the devel branch must be set, the devel branch must exist,
    and there must be a valid package.xml in the upstream devel branch.
  <version>
    This will be the version used.
    It must be updated for each new upstream version.
  [':{auto}']:
Upstream Devel Branch:
  <vcs reference>
    Branch in upstream repository on which to search for the version.
    This is used only when version is set to ':{auto}'.
  ['master']: galactic 
```
This did not resolve the issue:

https://build.ros2.org/job/Gpr__rclc__ubuntu_focal_amd64/50/console

Any idea, what I could do?

Anything else I could do?

JanStaschulat, 2021-06-09:
Any feedback?

nuclearsandwich, 2021-06-09:
@JanStaschulat I don't see how the PR job failures are related to the bloom release?

Based on your reported bloom release operations you have done everything as I would have done it and the _release_ builds for rclc on Galactic are currently succeeding on both [amd64](https://build.ros2.org/job/Gbin_uF64__rclc__ubuntu_focal_amd64__binary/) and [armv8](https://build.ros2.org/job/Gbin_ufv8_uFv8__rclc__ubuntu_focal_arm64__binary/).

It does look like PR jobs have recently started failing for rclc and upon investigation it looks like it was a problem with the specific Jenkins agent that the jobs were being run on. By temporarily disabling that agent I was able to re-trigger an example PR's CI: https://github.com/ros2/rclc/pull/100#issuecomment-857985148

That build has succeeded and I'll be retiring the Jenkins agent which was exhibiting the problem.

JanStaschulat, 2021-06-10:
@nuclearsandwich Thanks for re-triggering the ci-job. The build is now successful. I created a dummy new PR https://github.com/ros2/rclc/pull/101 which also runs successfully at the build farm. Thanks a lot. 

I wonder, if this problem occurs again in the future, could I re-start the build job also? Or should I report this problem here at rosdistro again? I am asking this question, because just creating a new pull request to galactic (with trivial change) did not resolve the error.

nuclearsandwich, 2021-06-10:
> However, what would I do, if this problem occured again?

We recently set up a repository specifically for fielding issue reports with Open Robotics infrastructure at [osrf/infrastructure](https://github.com/osrf/infrastructre) and I think that reporting problems like this there is reasonably valid. An issue there would help us track and mitigate the immediate issue and can reference any long running issues with our ros_buildfarm or cookbook deployment logic for trying to reduce the incidence of similar problems over time. The issue template in that repository will also recommend checking [ROS Answers](https://answers.ros.org) for similar problems before opening an issue in case there is a self-service solution and that should still be done as a first step, however there was not a self-service solution in this case. Thanks for raising the issue and hopefully by providing a directed place for these issues we'll be able to identify and respond more promptly.

closed 2021-06-10 by nuclearsandwich

JanStaschulat, 2021-06-11:
@nuclearsandwich Great to have a dedicated repository to issue such error reports. 

You might want to correct the link to `osrf/infrastructure` in our post. It  has a typo: `https://github.com/osrf/infrastructre`

nuclearsandwich, 2021-06-14:
> You might want to correct the link in our post. It has a typo: `https://github.com/osrf/infrastructre`

Thanks, I could have sworn I checked each link but that's how it goes I guess.