ros/rosdistro #30114: Debugging librealsense2 for Galactic Build Fail

doronhi, 2021-07-06:
The Galactic build of librealsense2 fails with an error message:
"
CMake Error at CMake/lrs_options.cmake:12:
09:02:49   Parse error.  Expected a newline, got identifier with text "option".
"
see [example build log](https://build.ros2.org/job/Gbin_uF64__librealsense2__ubuntu_focal_amd64__binary/6/consoleFull)

I assume the package described in the [distribution.yaml](https://github.com/ros/rosdistro/blob/9cc7b9d7ba865d89ae83dd02a842d409c01c047c/galactic/distribution.yaml#L1202) is [this](https://github.com/IntelRealSense/librealsense2-release/releases/tag/release%2Fgalactic%2Flibrealsense2%2F2.48.0-1). I downloaded the [source code.zip](https://github.com/IntelRealSense/librealsense2-release/archive/refs/tags/release/galactic/librealsense2/2.48.0-1.zip) but the file CMake/lrs_options.cmake at line 12 seems perfectly ordinary.
Running `cmake ../ -DCMAKE_BUILD_TYPE=None -DCMAKE_VERBOSE_MAKEFILE=ON -DCMAKE_AUTOGEN_VERBOSE=ON` also seems to work fine.
Am I donwloading a wrong version?
I am pretty sure it has something to do with the patches.
Is there a method to duplicate the process locally so I could see what is the problematic file looks like at the time CMake is called?
Any other ideas or suggestions for starting points for debugging?
Thanks

nuclearsandwich, 2021-07-13:
A short preface: In the future, this type of issue report / question is probably best posted to [ROS Answers](https://answers.ros.org). I may end up re-creating the issue there so the answer below can serve as a template / case study for investigating strange build issues.

My first thought is to suspect some kind of line-ending tomfoolery like a stray carriage return.
You've got the correct release, but for a more accurate reproduction of what the build farm has collected I would first look at the debian/* tag rather than the release/* tag.  The difference is that the debian/* tag is what the Ubuntu sourcedeb job will fetch in order to build the .deb package as that tag includes the debian metadata overlaid onto the release. Both the release "phase" and debian overlay "phase" in Bloom have the opportunity to patch files so it could be a problem introduced in the Debian phase. However, I downloaded the [source at debian/ros-galactic-librealsense2_2.48.0-1_focal](https://github.com/IntelRealSense/librealsense2-release/releases/tag/debian%2Fros-galactic-librealsense2_2.48.0-1_focal) and looked at it in xxd and I don't see anything funky about line 12. The next thing to check is if there's different behavior between downloading the archive and a git clone of the repository which is how the [sourcedeb job](https://build.ros2.org/job/Gsrc_uF__librealsense2__ubuntu_focal__source/) fetches the sources for the first time.

```
git clone --branch debian/ros-galactic-librealsense2_2.48.0-1_focal --depth 1 --no-single-branch https://github.com/IntelRealSense/librealsense2-release.git
```

However again, I don't find anything out of the ordinary there.
So let's work backwards from the binarydeb job.
We can reproduce the job itself on a Linux system with Docker configured using the `generate_release_script.py` command from [ros_buildfarm](https://github.com/ros-infrastructure/ros_buildfarm)

```
mkdir /tmp/librealsense2-galactic
generate_release_script.py https://raw.githubusercontent.com/ros2/ros_buildfarm_config/ros2/index.yaml galactic default librealsense2 ubuntu focal amd64 > /tmp/librealsense2-galactic/run.sh
cd /tmp/librealsense2-galactic
sh run.sh
```

This will run a script which performs _both_ the sourcedeb and binarydeb jobs locally via docker. If it does work then something went wrong on the Jenkins host and we may need to rebuild the debian source tarball.

Building locally worked for me. So let's pull down the [orig.tar.gz](http://repo.ros2.org/ubuntu/building/pool/main/r/ros-galactic-librealsense2/) file packed by the sourcedeb job. And if I untar that I find something very weird indeed!

```diff
 diff -u librealsense2-release/CMake/lrs_options.cmake ros-galactic-librealsense2-2.48.0/CMake/lrs_options.cmake
--- librealsense2-release/CMake/lrs_options.cmake       2021-07-13 10:47:21.389278815 -0700
+++ ros-galactic-librealsense2-2.48.0/CMake/lrs_options.cmake   2021-07-01 05:02:24.000000000 -0700
@@ -1,4 +1,15 @@
-option(ENABLE_CCACHE "Build with ccache." OFF)
+option(ENABLE_CCACHE "Build with ccache." ON)
+option(BUILD_WITH_CUDA "Enable CUDA" OFF)
+option(BUILD_GLSL_EXTENSIONS "Build GLSL extensions API" ON)
+option(BUILD_WITH_OPENMP "Use OpenMP" OFF)
+option(ENABLE_ZERO_COPY "Enable zero copy functionality" OFF)
+option(BUILD_WITH_TM2 "Build with support for Intel TM2 tracking device" ON)
+option(BUILD_EASYLOGGINGPP "Build EasyLogging++ as a part of the build" ON)
+option(BUILD_WITH_STATIC_CRT "Build with static link CRT" ON)
+option(HWM_OVER_XU "Send HWM commands over UVC XU control" ON)
+option(COM_MULTITHREADED "Set OFF to initialize COM library with COINIT_APARTMENTTHREADED (Windows only)" ON)
+option(BUILD_SHARED_LIBS "Build shared library" ON)
+option(BUILD_UNIT_TESTS "Build LibCI unit tests. If enabled, additional test data may be downloaded" OFF)option(ENABLE_CCACHE "Build with ccache." OFF)
 option(BUILD_WITH_CUDA "Enable CUDA" OFF)
 option(BUILD_GLSL_EXTENSIONS "Build GLSL extensions API" ON)
 option(BUILD_WITH_OPENMP "Use OpenMP" OFF)
@@ -34,6 +45,41 @@
 option(BUILD_NETWORK_DEVICE "Build Network Device support" OFF)
 option(FORCE_LIBUVC "Explicitly turn-on libuvc backend - deprecated, use FORCE_RSUSB_BACKEND instead" OFF)
 option(FORCE_WINUSB_UVC "Explicitly turn-on winusb_uvc (for win7) backend - deprecated, use FORCE_RSUSB_BACKEND instead" OFF)
+option(ANDROID_USB_HOST_UVC "Build UVC backend for Android - deprecated, use FORCE_RSUSB_BACKEND instead" OFF)
+option(CHECK_FOR_UPDATES "Checks for versions updates" ON)
+option(BUILD_WITH_CPU_EXTENSIONS "Enable compiler optimizations using CPU extensions (such as AVX)" ON)
+set(UNIT_TESTS_ARGS "" CACHE STRING "Command-line arguments to pass to unit-tests-config.py, e.g. '-t <tag> -r <regex>'")
+#Performance improvement with Ubuntu 18/20
+if(UNIX AND (NOT ANDROID_NDK_TOOLCHAIN_INCLUDED))
+    option(ENABLE_EASYLOGGINGPP_ASYNC "Switch Logger to Asynchronous Mode (set OFF for Synchronous Mode)"  ON)
+else()
+    option(ENABLE_EASYLOGGINGPP_ASYNC "Switch Logger to Asynchronous Mode (set OFF for Synchronous Mode)" OFF)
+endif()
+
+option(BUILD_LEGACY_LIVE_TEST "Build the live-test.exe test used in Travis. If enabled, additional test data may be downloaded" OFF)
+option(BUILD_INTERNAL_UNIT_TESTS "Test package for components under librealsense namespace, requires BUILD_SHARED_LIBS=OFF and BUILD_UNIT_TESTS=ON" OFF)
+option(BUILD_EXAMPLES "Build examples (not including graphical examples -- see BUILD_GRAPHICAL_EXAMPLES)" ON)
[... cut ...] he current sources and did properly observe the "correct" state of the file:

```
11:00:12 dpkg-source: info: building ros-galactic-librealsense2 using existing ./ros-galactic-librealsense2_2.48.0.orig.tar.gz

11:00:13 dpkg-source: info: local changes detected, the modified files are:
11:00:13  source/CMake/lrs_options.cmake
```

Due to the way debian source packaging works the orig.tar.gz file will remain unchanged and the file will be modified when fetched via the debian source build pipeline.

doronhi, 2021-07-14:
Thank you so much, both for looking into it so thoroughly and for the excellent description of the different jobs the build farm does. It is a great resource for debugging future releases.
Please let me know if you want me to re-create it in ROS Answers.
If I understand correctly the package will now be built and released, although we are not quite sure how I managed to mess it up so as to confuse git.
Assuming so, the patches of this version may very well be the base for the next version and so I'm afraid the issue will return very soon, with the next librealsense2 release.
The lrs_options.cmake is a file with patches that I tried to migrate from foxy and it is possible I somehow messed it up with the bloom-release process.
I released the same librealsense2 version again at #30152 (this time as 2.48.0-**2**), hopefully without any patch/code duplications. Is it possible for me to test it using  `generate_release_script.py` and see if it's worth merging instead of the current version (2.48.0-**1**)?

nuclearsandwich, 2021-07-14:
> although we are not quite sure how I managed to mess it up so as to confuse git.
Assuming so, the patches of this version may very well be the base for the next version and so I'm afraid the issue will return very soon, with the next librealsense2 release.
The lrs_options.cmake is a file with patches that I tried to migrate from foxy and it is possible I somehow messed it up with the bloom-release process.

Can you share the steps that you took to apply patches for Foxy and Galactic with as much detail as you recall?
Did you manually update or manipulate any of the tags in the release repository?

As it happens, updating bloom's docs to better explain the release and platform patch workflows is on my TODO list for this Summer so this could be its own fruitful case study.

> If I understand correctly the package will now be built and released

Yes! This was apparently the only build blocker so the package is currently in the [ROS 2 Testing repository](https://docs.ros.org/en/ros2_documentation/galactic/Installation/Prerelease-Testing.html#debian-testing-repository) if you want to set that up and manually test the binary. The [next Galactic sync](https://discourse.ros.org/t/preparing-for-galactic-sync-and-patch-release-2021-07-08/21273) is coming and that will include this release of librealsense2. The status of packages in ROS repositories are viewable on the repository status pages like [this one for Galactic on Ubuntu AMD64](http://repo.ros2.org/status_page/ros_galactic_default.html?q=librealsense2).

> I released the same librealsense2 version again at #30152 (this time as 2.48.0-**2**), hopefully without any patch/code duplications. Is it possible for me to test it using `generate_release_script.py` and see if it's worth merging instead of the current version (2.48.0-**1**)?

Possible: yes. However it is far from straightforward since the generated build script reads a ros_buildfarm_config which references the public rosdistro index so without creating your own custom rosdistro and ros_buildfarm_config there's not an easy way to "mock" a release.

Comparing the two tags locally, the only content differences are the package revision number and changelog date. So the source contents are identical and I don't think releasing 2.48.0-2 is necessary.

doronhi, 2021-07-15:
I am so frustrated that I already closed the terminals since, so now I need to recreate the steps from memory and personal documentation:
First - I updated the foxy version.
`bloom-release --ros-distro foxy --track foxy librealsense2 --edit`
I got this message: (As expected because the lrs_options.cmake file was changed and the patches no longer match)
```
error: CMake/lrs_options.cmake: patch does not apply
>>> Resolve any conflicts and when you have resolved this problem run 'git am --resolved' and then exit the shell using 'exit 0'. <<<
```
Then I did:
`gedit CMake/lrs_options.cmake` - Manually modified the file.
Then:
```
git add CMake/lrs_options.cmake
git am --resolved
exit 0
```

Second - I moved on to release the galactic version. There is no galactic version at this point so I migrate the patches from foxy:
`bloom-release --ros-distro galactic --track galactic librealsense2 --new-track`

When asked to migrate patches from another distribution I opened another terminal and did:
```
cd /tmp/tmp3cAQzd/
git-bloom-release galactic
```
Here it had some troubles because without patches it is looking for "catkin" but we pressed on. Then:
```
git checkout patches/release/foxy/librealsense2
mkdir ~/temp/patches
cp *.patch ~/temp/patches
git checkout release/galactic/librealsense2
git am ~/temp/patches/*
```
I don't remember but I guess the patches didn't apply because the fixed foxy version is not merged at this point, so I probably fixed the lrs_options.cmake, the way I did for foxy:
```
gedit CMake/lrs_options.cmake
git add CMake/lrs_options.cmake
git am --resolved
```
Then I carried on according to my list:
```
git-bloom-patch export
```
Now I returned to the original terminal and typed "Y" - continue.
Here I just carried on with the process. I don't remember pushing any tags manually.
I hope it helps in the documentation process you are engaged in.

doronhi, 2021-07-15:
By the way, do you still think these types of questions should be asked in the ROS answers forum? Why?
I asked another release-related question in ROSanswers [here](https://answers.ros.org/question/382540/realsense2_camera-galactic-release-fails-on-ros-buildfarm/) Do you think it belongs in this repository instead?

doronhi, 2021-07-21:
For some reason version 2.48.0-2, from #30152 build is successful: https://build.ros2.org/job/Gsrc_uF__librealsense2__ubuntu_focal__source/

Although we don't really understand what went wrong, I feel like this issue can be closed.
Thanks for all your help.

clalancette, 2021-11-08:
Based on the last comment, closing this.

closed 2021-11-08 by clalancette