ros/rosdistro #30138: python-serial missing for ubuntu focal?

flynneva, 2021-07-08:
not sure if this is actually missing or purposefully omitted, but it doesnt look like `python-serial` has a ubuntu `focal` line item yet.

want me to make a PR to add it in?

nuclearsandwich, 2021-07-08:
Since focal does not have python2 packages, I don't think it makes sense to update the `python-serial` key for Focal. `python3-serial` is generically defined for Ubuntu and so packages in Noetic can use conditional dependencies based on the ROS_PYTHON_VERSION to depend on either python-serial or python3-serial https://github.com/ros/rosdistro/blob/6a23f467bd2392ef46be34f19ac4805d4be7e0eb/rosdep/python.yaml#L7642

flynneva, 2021-07-08:
That makes a lot of sense, thanks.

closed 2021-07-08 by flynneva