ros/rosdistro #30351: swri-robotics/marti_common packages listed in galactic but repos do not exist

Tobias-Fischer, 2021-07-29:
Hiya,

In our RoboStack efforts we encountered an issue where `marti_common` packages are supposedly released into galactic - see https://github.com/ros/rosdistro/blob/76a7ae95c82edd2eeb5d02bb65a279aa34c5cefb/galactic/distribution.yaml#L1345-L1372

However, the release repo (https://github.com/swri-robotics-gbp/marti_common-release) does not contain any branches starting with `release/galactic/`

I'm not sure what the proper way of going forward is - ideally I guess properly blooming the `marti_common` packages, or temporarily removing them from the galactic/distribution.yaml?

/cc @wolfv @traversaro @nuclearsandwich

cottsay, 2021-07-29:
The buildfarm is ignoring this repo because it doesn't have a `version` value under the `release` section. Can you update your automation to act similarly?

cottsay, 2021-07-29:
To further clarify - we've used this mechanism in the past to disable or retract a package while maintaining information in the index that will make future releases easier. In this case, there was indeed a problem during the Bloom run that we didn't catch during review. While it would be good for the maintainers to fix this package, there will probably be future instances of this pattern, so you should probably handle it appropriately.

Tobias-Fischer, 2021-07-30:
Ah, I see. That's good to know. We indeed skip this package at the moment, so it's all good. Your description makes sense; it's a non-issue then. Thanks @cottsay!

closed 2021-07-30 by Tobias-Fischer