ros/rosdistro #30425: micro_ros_setup No definition of [python3-vcstool] for OS [osx]

srliu, 2021-08-07:
I'm tying to install a micro-ROS development on my Macbook (Big Sur11.4) and have run into issues after the following steps:
`# Update dependencies using rosdep
sudo rosdep update
rosdep install --from-path src --ignore-src -y
`

I end up with the error:
`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
micro_ros_setup: No definition of [python3-vcstool] for OS [osx]
`
Looking at file `ros/rosdistro/rosdep/python.yaml `, it appears there is indeed no key for `osx`:
`
python3-vcstool:
  alpine: [vcstool]
  debian:
    pip:
      packages: [vcstool]
  fedora: [python3-vcstool]
  gentoo:
    pip:
      packages: [vcstool]
  macports:
    pip:
      packages: [vcstool]
  ubuntu: [python3-vcstool]
`

My python version is Python 3.9.2. ROS2 version is Foxy.

Can anyone help with this issue?

jacobperron, 2021-08-09:
ROS 2 installation instructions for macOS avoid using rosdep to install dependencies. The vcstool dependency is installed via pip: https://docs.ros.org/en/foxy/Installation/macOS-Development-Setup.html#install-prerequisites

I'm not familiar enough with micro-ROS to know if macOS. This sounds like an issue that is better to ask over at the micro-ROS org (maybe here: https://github.com/micro-ROS/micro_ros_setup).

You might be able to workaround the issue by installing the equivalents from pip for any missing dependencies. Note, you can add the `-r` option to the rosdep install line to continue on error (so that you can at least get all available dependencies from the system): `rosdep install --from-paths src --ignore-src -y -r`

Since I don't think this is an issue with rosdistro, I'm going to close this ticket.

closed 2021-08-09 by jacobperron

srliu, 2021-08-10:
thank you, jacobperron!