ros/rosdistro #30904: Backport absl Ubuntu pacakges?

tylerjw, 2021-10-02:
Would it be possible to add a couple of packages that are released into newer versions of Ubuntu to the ROS focal ppa?  Specifically, the absl library is released for groovy and I would really like to use it on focal.  Here is the source and library package in Ubuntu Groovy:

https://packages.ubuntu.com/groovy/libabsl20200225
https://packages.ubuntu.com/groovy/libabsl-dev

If this is the wrong place for this please let me know.  I also asked here on launchpad to see if maybe Ubuntu could do it themselves:  https://answers.launchpad.net/ubuntu/+source/abseil/+question/698936

Here is the library I'd like to use: https://github.com/abseil/abseil-cpp

clalancette, 2021-10-05:
This one is tricky.

What we don't typically allow is a package in the ROS distribution to override a system package on Ubuntu; that is a recipe for ABI and dynamic linking disasters.

In this case, the library is not yet available in our supported platforms (Bionic on Focal, essentially).  On those grounds, we could allow the library in, but we have to consider what is going to happen when we switch to Ubuntu 22.04 (where presumably abseil will be available from the OS).

With that in mind, it seems to me the best way forward here is to make a `abseil_vendor` package, similar to what we are doing in https://github.com/ros2/console_bridge_vendor .  That is, the CMake logic first tries to find the installed library; if that is found, it just uses that.  If that is not found, then it downloads, builds, and installs a local copy.  Downstream packages depend on the vendor package, which directs them on how to find either the OS library or the locally-built copy.  With that in place, this would work for both Ubuntu 20.04 (where we would build the library), and for Ubuntu 22.04 (where we would use the system copy of the library).

Does that make sense?

tylerjw, 2021-10-05:
Thank you for the detailed response.  I will attempt to make this work.  Long term though as absl will be released for future versions of Ubuntu what will happen to the vendor package?

clalancette, 2021-11-08:
Given that this is more of a question than a bug report or request for changes, I'm going to close this out.  @tylerjw feel free to reopen if you need to discuss further.

closed 2021-11-08 by clalancette