ros/rosdistro #31215: rosdistro entry requiring PPA?

SteveMacenski, 2021-11-12:
We're looking to add a new package to Nav2 that needs `xtensor` which has public releases for 18.04/21.10 but not 20.04. There is, however, a PPA with the package available to use (see below). Is it possible to get rosdistro entries requiring adding a new PPA? I do not believe I've seen documentation about this previously.  

https://answers.launchpad.net/ubuntu/+source/xtensor/+question/699443

jacobperron, 2021-11-22:
I'm pretty sure ROS infrastructure doesn't support requiring a specific PPA for a rosdep rule.

@nuclearsandwich may have a more concrete answer.

SteveMacenski, 2021-11-22:
I assumed so, but figured it would be good to ask :smile:

mikeferguson, 2021-11-22:
I know in the past @tfoote helped me import a PPA-built version of ceres-solver into packages.ros.org when the built was broken in a particular Ubuntu distro.

hidmic, 2021-12-15:
@nuclearsandwich shall we close? AFAIK there's no plan to support custom PPAs in the foreseeable future.

tfoote, 2022-01-05:
Yes, we explicitly require that content is either in the upstream or the ROS sources: https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#ubun

If it's already released into 18.04 and 21.10 we can probably assist getting it backported  20.04 with the debian/ubuntu maintainers if it's still of interest. Email me direcly and we can start a thread looking to get it released upstream into focal.

closed 2022-01-05 by tfoote

nuclearsandwich, 2022-01-05:
Thanks @tfoote  for jumping in and apologies upthread for not fielding the mentions. I've been swamped and am in the process of clearing out direct mentions before declaring GitHub bankruptcy so that I'm in a better place going into 2022.

SteveMacenski, 2022-01-10:
OK, will do when the time comes!