ros/rosdistro #31439: Repo check test is checking the wrong lines

ivanpauno, 2021-12-10:
I have seen two differece instances of the issue:

- It checks the correct lines, but the error is reported in an incorrect line: https://github.com/ros/rosdistro/pull/30565/files.

<details>
<summary>screenshot</summary>

[image]

[image]

</details>

- It checks the incorrect lines (IIUC, it should only be checking added lines): https://github.com/ros/rosdistro/pull/31402/files

<details>
<summary>screenshot</summary>

[image]

[image]

</details>

Both thinks are likely different issues though.

The first issue is maybe related to my comment [here](https://github.com/ros/rosdistro/pull/30694#issuecomment-919237902).
The second one is likely related to the diff having some extra context lines.
(those are just wild guesses)

ivanpauno, 2021-12-10:
cc @cottsay

cottsay, 2021-12-14:
Alright, I've reproduced two issues that the current implementation is facing.
1. The HEAD in the checked out repository [can be a merge commit](https://github.com/actions/checkout#checkout-pull-request-head-commit-instead-of-merge-commit) with `master`, which is why the line numbers can get off.
2. If `master` has a commit which removes rules from rosdep but the PR branch does not, the diff shows those lines as being new.

Based on this, I think we need to:
a. Introduce a third ref in the workflow to represent the PR's tip. The local checkout should continue to use the merge so that the testing infrastructure is up-to-date. So the three important refs are the merge, the PR tip, and the target tip.
b. Use git's triple-dot to indicate that we only want to see "new" commits present in the PR but not in master, which should avoid (2).