ros/rosdistro #32030: error installing new release of cnpy

PeterMitrano, 2022-02-03:
Trying to install the latest noetic release of cnpy gives the following error:

```Preparing to unpack .../ros-noetic-cnpy_0.0.7-3focal.20211207.001108_amd64.deb ...
Unpacking ros-noetic-cnpy (0.0.7-3focal.20211207.001108) ...
dpkg: error processing archive /var/cache/apt/archives/ros-noetic-cnpy_0.0.7-3focal.20211207.001108_amd64.deb (--unpack):
 trying to overwrite '/opt/ros/noetic/.catkin', which is also in package ros-noetic-catkin 0.8.10-1focal.20210423.221558
Errors were encountered while processing:
 /var/cache/apt/archives/ros-noetic-cnpy_0.0.7-3focal.20211207.001108_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I am the package's maintainer, but I have no idea why this error happens or how my changes caused it. Here's the diff between 0.0.6 which worked and 0.0.7 which did not.

```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 40706a0..9ea2743 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -3,13 +3,15 @@ if (COMMAND cmake_policy)
     cmake_policy(SET CMP0003 NEW)
 endif (COMMAND cmake_policy)
 
-project(CNPY)
+project(cnpy)
 
 set(CMAKE_CXX_STANDARD 11)
 set(CMAKE_WINDOWS_EXPORT_ALL_SYMBOLS ON)
 
 find_package(ZLIB REQUIRED)
+find_package(catkin REQUIRED)
 
+catkin_package()
 
 add_library(cnpy SHARED cnpy.cpp)
 target_include_directories(cnpy PUBLIC
diff --git a/package.xml b/package.xml
index 148b8d3..ede7165 100644
--- a/package.xml
+++ b/package.xml
@@ -1,7 +1,7 @@
 <?xml version="1.0"?>
 <package>
   <name>cnpy</name>
-  <version>0.0.6</version>
+  <version>0.0.7</version>
   <description>library to read/write .npy and .npz files in C/C++</description>
   <maintainer email="pmitrano@umich.edu">Peter Mitrano</maintainer>
   <license>MIT</license>
@@ -9,7 +9,7 @@
   <author>Carl Rogers</author>
   <author>Peter Mitrano</author>
 
-  <buildtool_depend>cmake</buildtool_depend>
+  <buildtool_depend>catkin</buildtool_depend>
 
   <build_depend>cmake</build_depend>
   <build_depend>zlib</build_depend>
```

I was sure using catkin as tool depend was the right thing, but maybe not. Can anyone help explain what's broken and how to fix it?

tfoote, 2022-03-03:
Your package appears to be installing a .catkin file outside of it's appropriate directory. I believe that this default behavior is usually disabled in the default release template. Such as here: https://github.com/ros-gbp/geometry2-release/blob/3b13f39e9a653d48eb513d1fa6f7a445dc3023ec/debian/rules#L30 

However your package doesn't have this callout: https://github.com/PeterMitrano/cnpy-release/blob/d38dccd77268603a55f22542e00fa2f580282fe2/debian/rules#L32-L34 

I think that your release template was likely patched because your previous project was a pure cmake package instead of the default catkin package. And consequently when you now switch to use catkin that patch needs to be rolled back.

cottsay, 2022-03-03:
Hi, @PeterMitrano. I'm moderately confident that the template issues that @tfoote is describing are caused by the incorrect `build_type` declared in [your package manifest](https://github.com/PeterMitrano/cnpy/blob/445491fc0b0d7b5e0e4175bf3e425259f68f7494/package.xml#L17).

According to [the specifications](https://ros.org/reps/rep-0127.html#build-type), the `build_type` should either be explicitly set to `catkin` or omitted for Catkin packages.

PeterMitrano, 2022-03-03:
I thought that might be the case. I've removed that and pushed a new release, and we'll see if that fixes it.

https://github.com/ros/rosdistro/pull/32383

cottsay, 2022-03-03:
Looks like the template now has the expected CMake definition: https://github.com/PeterMitrano/cnpy-release/blob/a35b52d96be44583c4975d0504847d4e346cfad6/debian/rules#L30

I'm optimistic.

tfoote, 2022-03-04:
I've confirmed 0.0.8 is installable from testing. 

```
root@f4000f9cc97b:/# apt-cache policy ros-noetic-cnpy
ros-noetic-cnpy:
  Installed: (none)
  Candidate: 0.0.8-1focal.20220303.214014
  Version table:
     0.0.8-1focal.20220303.214014 500
        500 http://packages.ros.org/ros-testing/ubuntu focal/main amd64 Packages
root@f4000f9cc97b:/# apt-get install ros-noetic-cnpy
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  ros-noetic-cnpy
0 upgraded, 1 newly installed, 0 to remove and 113 not upgraded.
Need to get 62.9 kB of archives.
After this operation, 251 kB of additional disk space will be used.
Get:1 http://packages.ros.org/ros-testing/ubuntu focal/main amd64 ros-noetic-cnpy amd64 0.0.8-1focal.20220303.214014 [62.9 kB]
Fetched 62.9 kB in 0s (173 kB/s)           
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 68167 files and directories currently installed.)
Preparing to unpack .../ros-noetic-cnpy_0.0.8-1focal.20220303.214014_amd64.deb ...
Unpacking ros-noetic-cnpy (0.0.8-1focal.20220303.214014) ...
Setting up ros-noetic-cnpy (0.0.8-1focal.20220303.214014) ...
root@f4000f9cc97b:/# 
```

closed 2022-03-04 by tfoote