ros/rosdistro #32236: Next Melodic sync?

rhaschke, 2022-02-18:
I recently fixed some regressions:
- https://github.com/ros/rosdistro/pull/32144
- https://github.com/ros/rosdistro/pull/32142
@clalancette, do you plan a new sync anytime soon? Unfortunately, there are 3 regressions...

clalancette, 2022-02-18:
Hm, maybe.  There are a lot of regressions on Melodic right now:

* http://repositories.ros.org/status_page/ros_melodic_default.html?q=REGRESSION
* http://repositories.ros.org/status_page/ros_melodic_ubhf.html?q=REGRESSION

I'll see if I can improve any of that.

tfoote, 2022-03-03:
Pleaes use the release management channels for discussing syncs so that it's more visible to the community. https://discourse.ros.org/c/release/melodic/38

Since @clalancette is aware of this. I'm going to close this as it's not something that we track here.

closed 2022-03-03 by tfoote