ros/rosdistro #32417: Rolling on Focal

artivis, 2022-03-07:
Hello,

I understand that Rolling was migrated to Jammy in #32036 however since Rolling's [`distribution.yaml` only lists `jammy`](https://github.com/ros/rosdistro/blob/d4c7a7c49fbed4cbd94e0d1ca9a4139a792dc993/rolling/distribution.yaml#L11) as a release platform `rosdep` is now inoperative on Focal. I suppose I could work around that with the `--rosdistro` flag but I'm sure this will cause other issues in the future (e.g. packages that are only released on Jammy).

I'm not sure what a 'proper' fix would be. Maybe a new `rosdep` argument to fetch the distribution at a given commit? Any other idea?

tfoote, 2022-03-08:
You're taking advantage of an intermediate state right now where the focal packages still exist. Note that the packages built on focal are also expected to disappear and cannot be relied upon looking forward.

If you're building on focal you will have to do everything from source. And you likely want to be using a tool like `rosinstall_generator` instead of `rosdep`. The focal packages are not considered part of rolling anymore and there's no guarentees about correlation between them and the ongoing distro. (Including things like the default rmw potentially changing breaking all binary compatibility)

artivis, 2022-03-15:
I had not realized that Rolling is "expected to disappear" all together. This changes perspective, thanks for the pointers !

closed 2022-03-15 by artivis