ros/rosdistro #32471: Permission denied error building Python package using cmake

johnwason, 2022-03-17:
My package builds a fairly complicated Python module using SWIG and CMake, and uses a Python setup.py command to install. The install commands can be found here:

https://github.com/robotraconteur/robotraconteur/blob/868128446a89614a4626e8587de9ba2d5807d757/RobotRaconteurPython3/CMakeLists.txt#L129

This code works correctly on all other platforms, including building normal Ubuntu and Debian packages. However, on the Jenkins build I am seeing this error:

```
error: could not create '/opt/ros/noetic/lib/python3/dist-packages/RobotRaconteur': Permission denied
```

Here is the full build log:

https://build.ros.org/job/Nbin_uF64__robotraconteur__ubuntu_focal_amd64__binary/1/console

Why am I seeing this error? The generated rules file is setting `CMAKE_INSTALL_PREFIX` to `/opt/ros/noetic` so it is attempting to install to the correct location.

https://github.com/robotraconteur-packaging/robotraconteur-ros-release/blob/f6f6efe5088cdf24f0121dc2c08bd5880c41ea03/debian/rules#L33

Levi-Armstrong, 2022-03-19:
I believe it should be installing into this location `debian/ros-noetic-robotraconteur/opt/ros/noetic/lib/python3/dist-packages/RobotRaconteur` since it is creating a debian.

johnwason, 2022-03-19:
I think that debuild is supposed to use something like `fakechroot` during the build so the installation location `/opt/ros/noetic` would be correct inside the chroot.

Levi-Armstrong, 2022-03-22:
I believe the issues maybe this `--root=/` in the command below.

```
execute_process(COMMAND ${PYTHON3_EXECUTABLE} setup.py install -f
--root=/ --prefix=${CMAKE_INSTALL_PREFIX} ${SETUPTOOLS_ARG_EXTRA} --single-version-externally-managed WORKING_DIRECTORY ${CMAKE_BINARY_DIR}/$\{RR_BUILD_CONFIG_DIR\}/Python3)")
```

Levi-Armstrong, 2022-03-22:
The `--prefix=${CMAKE_INSTALL_PREFIX}` should be a full path to the location to install and I image that the `--root=/` is taking priority.

johnwason, 2022-03-23:
I see the Python files being installed correctly in the log on Jenkins, but I don't see a link to download the package.

https://build.ros.org/job/Nbin_uF64__robotraconteur__ubuntu_focal_amd64__binary/2/console

closed 2022-04-04 by johnwason